RobotLocomotion/drake #22602: declare sparsity patterns for nonlinear constraints in KinematicTrajectoryOptimization

RussTedrake, 2025-02-09:
As discussed with @hongkai-dai ... we can make the new formulation of the constraints much more efficient (e.g. for Snopt) using [`SetSparsityPattern`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1solvers_1_1_evaluator_base.html#a604f98ac54b2b64e7dbe46efb31e9af1).

hongkai-dai, 2025-03-16:
Looking at the code in KinematicTrajectoryOptimization, I think we have already made some of the constraints sparse. For example in AddPathPositionConstraint(), we only select the active control points in https://github.com/RobotLocomotion/drake/blob/3d40a0fbaefca928d10832e96cb1ae20cdcfd3f8/planning/trajectory_optimization/kinematic_trajectory_optimization.cc#L386-L401, so this constraint should only depend on the active control points.

For the other nonlinear constraints, such as `AddVelocityConstraintAtNormalizedTime`, I believe we could improve the sparsity, when it calls `bspline_.EvaluateLinearInControlPoints()` function to compute a_pos and a_vel in https://github.com/RobotLocomotion/drake/blob/3d40a0fbaefca928d10832e96cb1ae20cdcfd3f8/planning/trajectory_optimization/kinematic_trajectory_optimization.cc#L430-L438. These `a_pos` and `a_vel` should be sparse, that the position and velocity of bspline shouldn't depend on all of the control points, but we are treating these `a_pos` and `a_vel` as dense matrix, and the returned constraint should depend on just a subset of the control points, but our implementation made it depend on all control points.

@RussTedrake is this consistent with your understanding?

RussTedrake, 2025-03-16:
Yes, that's my understanding.  Thanks!

PR #22912 (2025-04-22): Set sparsity pattern for KinematicTrajectoryOptimization effort bounds.
Resolves #22602

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22912)
closed 2025-04-22 by sammy-tri