RobotLocomotion/drake #22616: Rendering engine must be added to scene graph before building diagram

marip8, 2025-02-12:
### What happened?

It appears that rendering engines must be added to the scene graph before the diagram which contains the scene graph is built. If not, some system (I think it's the `RgbdSensor`) raise the following exception: `RuntimeError: No renderer exists with name: 'camera'`.

This is not especially problematic for me, but it does not appear to be documented anywhere that this is the case. I'm not sure if this is a bug or the intended functionality, but it would be helpful for this to be more clearly documented.

Here is some pseudo-code for re-creating this issue.

```python

from pydrake.all import RobotDiagramBuilder, MakeRenderEngineGl, RenderEngineGlParams, DiagramBuilder

rdb = RobotDiagramBuilder(time_step=0)
rdb.AddModelsFromURL(...)

# Add a renderer to the scene graph
# Results in correct behavior because this is called before `rdb.Build()`
rdb.scene_graph().AddRenderer('camera', MakeRenderEngineGl(RenderEngineGlParams())

robot = rdb.Build()

# Create a higher level diagram with other systems
builder = DiagramBuilder()
builder.AddNamedSystem('robot', robot)

# Add a renderer to the scene graph here instead
# Technically succeeds, but ultimately results in **incorrect** behavior because this is called after `rdb.Build()` where scene graph is built into robot diagram
# robot.scene_graph().AddRenderer('camera', MakeRenderEngineGl(RenderEngineGlParams())

# The following assert also succeeds, even though the end result at `ForcedPublish` is an exception that says no renderer by the same name can be found
# assert robot.scene_graph().HasRenderer('camera')

# Add a camera
...

# Add a system that connects to the camera 
...

# Connect subsystems together
...

# Build the diagram
diagram = builder.Build()

# Publish to trigger image generation
context = diagram.CreateDefaultContext()
diagram.ForcedPublish(context)
```

### Version

v1.36.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

pip install drake

### Relevant log output

```shell

```

ggould-tri, 2025-02-18:
Assigning component `geometry perception`; assigning @zachfang  for triage.

zachfang, 2025-02-24:
Thanks for reporting. I will take a look, try to reproduce the issue locally, and circle back this week.

zachfang, 2025-05-08:
Thanks for the update. I will close this issue for now, but you can reopen if you still encounter issues.

closed 2025-05-08 by zachfang