RobotLocomotion/drake #22617: Add missing python bindings for MeshcatAnimation

yangwill, 2025-02-12:
**Is your feature request related to a problem? Please describe.**
When I use the MeshcatVisualizer from pydrake and I want to only record a short segment of the visualizer to a MeshcatAnimation, the MeshcatAnimation starts at time = 0 instead of the Simulator time when I call `StartRecording()`.

**Describe the solution you'd like**
I see that there's a `set_start_time()`[function](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_meshcat_animation.html#ad6602e33d225fb38d213f5684cb18813)  in MeshcatAnimation to configure the start time, but there is no corresponding python binding in pydrake. I would like the c++ function for MeshcatAnimation to have corresponding python bindings to have access to the features in MeshcatAnimation.
 
**Describe alternatives you've considered**

**Additional context**

jwnimmer-tri, 2025-02-12:
It's probably worth doing both `start_time()` and `set_start_time()` at the same time.  Both seem to have been overlooked in #15937.

ggould-tri, 2025-02-18:
Assigning pydrake component owner @EricCousineau-TRI for traige.

PR #22650 (2025-02-19): Add missing bindings from MeshcatAnimation
Resolves #22617

+@xuchenhan-tri for both reviews, please.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22650)
commit 9ee48ce3 (2025-02-19): Add missing bindings from MeshcatAnimation (#22650)

Resolves #22617
closed 2025-02-19 by xuchenhan-tri