RobotLocomotion/drake #22624: Additional methods for the class `DiscreteTimeTrajectory`

wei-chen-li, 2025-02-13:
**Is your feature request related to a problem? Please describe.**
1. When a function returns a `DiscreteTimeTrajectory`, the `std::vector` used for its construction is unnecessarily copied.
2. There is no convenient method to sample from a `Trajectory` and generate a `DiscreteTimeTrajectory`.

**Describe the solution you'd like**
Implement the following three methods in the `DiscreteTimeTrajectory` class
```
DiscreteTimeTrajectory(const std::vector<T>& times, std::vector<MatrixX<T>>&& values, const double time_comparison_tolerance)

DiscreteTimeTrajectory(std::vector<T>&& times, std::vector<MatrixX<T>>&& values, const double time_comparison_tolerance)

static DiscreteTimeTrajectory<T> SampleFrom(const Tajectory<T>& traj, double time_period, double time_offset=0)
```

**Additional Considerations**

A potential issue with `SampleFrom` is handling cases where `traj.start_time() == -inf` or `traj.end_time() == inf`. Possible solutions include:

- Returning an empty trajectory
- Throwing an exception
- Defining a reasonable default start time and end time

Feedback on the preferred behavior for these edge cases would be appreciated.

cohnt, 2025-02-13:
Wouldn't the [`Trajectory::vector_values`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1trajectories_1_1_trajectory.html#a7607cc5c56514557ed9441479ae70e71) method do what you need? I guess this doesn't work for trajectories that are matrix-value yet.

wei-chen-li, 2025-02-17:
`Trajectory::vector_values` is sufficient for my needs, thanks.

PR #22653 (2025-03-05): Python bindings for DiscreteTimeTrajectory class
Also resolves #22624.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22653)
closed 2025-02-20 by wei-chen-li