RobotLocomotion/drake #22630: `MapQddotToAcceleration` and friends

RussTedrake, 2025-02-13:
[In kinematic trajectory optimization, we currently don't support qdot != v because we don't have MapQddotToAcceleration.](https://github.com/RobotLocomotion/drake/blob/5a637881e95c4d8ff768e746320420e0e606fa5b/planning/trajectory_optimization/kinematic_trajectory_optimization.cc#L244-L245)

Request:
- `MultibodyPlant::MapQddotToAcceleration()`
- `MultibodyPlant::MapAccelerationToQddot()`
- bonus: 
  - `MultibodyPlant::MakeQddotToAccelerationMap()`
  - `MultibodyPlant::MakeAccelerationToQddotMap()`

sherm1, 2025-02-13:
The existing [MapQdotToVelocity()](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_system.html#a10eba167739bfe8e7e419b5c8ae41537) and related functions are methods of System (and thus Diagram) so that they can be invoked on a Diagram without having to dig out a MultibodyPlant (MultibodyPlant just implements the underlying virtual). 

It would be easiest just to add a local function to MultibodyPlant but nicer to add it at the System level with appropriate overrideable virtuals. WDYT?

mitiguy, 2025-02-13:
Just to clarify:  @sherm1  -- does it make sense to start by adding to MultibodyPlant?

sherm1, 2025-02-13:
@mitiguy I think so, since MbP has to implement it anyway even if we add the System virtual. @RussTedrake would users likely have direct access to the MultibodyPlant?

RussTedrake, 2025-02-13:
In my use case, I do have the `MultibodyPlant`. I think that's definitely sufficient for now.

jwnimmer-tri, 2025-03-16:
@sherm1 is this is Paul's currently assigned task, please add it to the project board at a row consistent with its relative priority.