RobotLocomotion/drake #22634: Add Convenience Constructor for Vector of Formulas

AlexandreAmice, 2025-02-14:
It would be nice to be able to call the following code blobs.
```
prog = MathematicalProgram()
x = prog.NewContinuousVariables(4, "x")
prog.AddLinearConstraint(x <= np.ones(4))
prog.AddLinearConstraint(x <= 1)
prog.AddLinearConstraint(x == np.ones(4))
prog.AddLinearConstraint(x == 1)
```

Currently, all four options fail because we don't support vectorized construction of `Formula` objects (i.e. one gets an error with just `x <= np.ones(4)`).

Once we have sugar for constructing vectors of `Formula`, one would also like to add
```
Binding<LinearConstraint> AddLinearEqualityConstraint(
      const Eigen::Ref<const MatrixX<symbolic::Expression>>& v,
      const Eigen::Ref<const Eigen::MatrixXd>& lb,
      const Eigen::Ref<const Eigen::MatrixXd>& ub)
```
to MathematicalProgram.

This makes reading code involving MathematicalPrograms easier to read. For example:
```
prog.AddLinearConstraint(x <= np.ones(4))
prog.AddLinearConstraint(x, -np.inf*np.ones(4), np.ones(4))

```
The first is nice and easy to read. The second requires me remembering which arguments should be variables vs bound, and noticing that I said `AddLinearConstraint` (so I have inequalities) vs `AddLinearEqualityConstraint`.

jdetweiler1, 2025-02-14:
Hi! I'd like to work on this issue. I’ll start implementing it today. Let me know if you have any guidance!

ggould-tri, 2025-02-18:
Assigning component `mathematical program`; assigning @hongkai-dai for triage.

murphp15, 2025-03-27:
Could I take this issue? I would love to start contributing to the project.

closed 2025-03-28 by AlexandreAmice

murphp15, 2025-03-28:
@AlexandreAmice  is there any tickets you could assign to me? I would like to start contributing to this project. 
I've watched Russ Tedrake's lectures online and read https://underactuated.mit.edu/index.html.