RobotLocomotion/drake #22639: Construct IRIS-ZO Parameterization for Mimic Joints

cohnt, 2025-02-17:
@sadraddini has requested this feature, and I'm planning to implement it myself soon. (Assigning to myself +@cohnt) I'm creating the feature request because I wanted to solicit feedback before I do so. cc @RussTedrake 

It seems like there are multiple ways to pull this off. One is to just have the user specify which joints are mimic joints. This seems ugly, but for simple setups (e.g. parallel jaw grippers), it would probably be the most convenient? We could also redo the parsing, but that seems very silly!

Another is to go into the MultibodyPlant and look for coupler constraints. But I'm worried this might turn into a footgun if the user has a discrete plant and isn't using SAP as the contact solver -- in that case, the mimic will be ignored, and the user could miss the warning. And in a more principled sense, it seems weird to make this (purely kinematic) method depend on what the user has selected as a solver for the contact dynamics in the plant.

However, inspecting the plant for the constraints also suggests the workflow of looking for weld constraints. This would be a much nicer workflow for IRIS regions in manipulation contexts, as the user could just weld free floating objects to make regions, instead of having to make an entirely separate set of directives.

Maybe the best option is to have two different setups?

1) Introspect the MultibodyPlant for coupler and weld constraints. Warn the user that discrete and not SAP will not see mimic joints.
2) Manually specify a list of mimic joints, along with their multipliers and offsets.

cohnt, 2025-04-01:
As I've been working on the PR for this, I've been finding myself basically writing mostly MultibodyPlant code and test cases. I can try this myself, but I'm not really sure I'm qualified to be writing code for MBP.

@jwnimmer-tri do you know if anyone at TRI has cycles to help out with this? (Tagging you because I can't actually remember who the MBP component owner is.)

cohnt, 2025-04-01:
> It seems to me it would be totally fair to throw in case the user asks IRIS-ZO to obtain constraints from the plant, but the plant was non-SAP. Nobody should be using anything else anyway, these days. The other modes are kept for backcompat, not to be used in new code.

Do you think it's appropriate to throw for continuous-time plants as well?

> We probably don't have bandwidth for IRIS support (unless you can convince @hongkai-dai) but if there is a more general missing API for MultibodyPlant, we could maybe take over that part. I'm not sure what kind of code you have in play, here.

The functionality would be to provide minimal coordinates in the scenario where the MBP has coupler constraints (either user-specified for from <drake:mimic> tags). If a plant only has coupler constraints, then we can uniquely write every valid configuration as y=Ax+b, where A is a rectangular matrix whose entries are the multipliers, b is a vector whose entries are the offsets, and x lives in the lower-dimensional space.

Right now, I'm getting a list of coupler constraints and traversing over them to construct the matrix and vector. Then I use that for the IRIS stuff downstream.

jwnimmer-tri, 2025-04-01:
> Do you think it's appropriate to throw for continuous-time plants as well?

Depending on the API contract, maybe.  But also, it doesn't matter?  I believe if the user tries to add constraints to a continuous-time plant, the plant will immediately throw.

> Right now, I'm getting a list of coupler constraints and traversing over them to construct the matrix and vector.

Is this the only "MultibodyPlant code and test cases" you're talking about?  That seems pretty small.  Anyway, possibly you should just share the code outline you have, and then people can weigh in on where/how to help, or what different API the MbP could offer.  That could be on `#dynamics` slack if that's easier.

jwnimmer-tri, 2025-04-02:
Makes sense.  In that case, I would actually suggest opening a draft PR with the general outline, and ping MbP experts to help validate / review the algorithm and suggest test cases.  You could ask on `#dynamics` or just ping Sherm & Joe & Alejandro.

jwnimmer-tri, 2025-04-02:
Having looked at the code now, I suggest that @sherm1 should be on point for offering advice on the code / whether the approach is valid.

sherm1, 2025-04-02:
I'm sorry Thomas I don't understand enough about what you're trying to do to provide any useful advice. If there are some MbP services that would make your life easier I'd be happy to discuss those. Possibly @joemasterjohn is more familiar with the application area as well as the mimic constraint internals. WDYT Joe?

jwnimmer-tri, 2025-04-02:
The question is whether the approach in `IrisZoOptions::CreateWithMimicJointsParameterization` is robust against all kinds of possible plant species, both today and as MbP evolves.  It has an API contract and unit tests, so that should be able to provide some leverage to understand the question.

sherm1, 2025-04-02:
> The question is whether the approach in IrisZoOptions::CreateWithMimicJointsParameterization is robust against all kinds of possible plant species, both today and as MbP evolves.

The answer to that is certainly no but I don't think it's the right question. The general problem of finding a set of independent coordinates for a constrained multibody system is very hard -- there are lots of constraint types, constraints can be redundant or ill conditioned, redundancy and conditioning can be configuration dependent, and there are arbitrary choices to be made about which coordinates are to be declared independent. But I don't think @cohnt is trying to solve that problem. So I'm thinking the right question is whether this serves the intended purpose when applied to the systems of actual interest to the author.

jwnimmer-tri, 2025-04-02:
That's fine.  In that case the predicate "systems of actual interest" is whatever the API contract of #22733 says it allows / supports.

So maybe the first step is to iterate on what that contract could/should say, and then second decide if the implementation is safe / sound?

The worlds "mimic joints" in the function makes me think that it *only* need to allow mimic joints, and any/all other kinds of constraints present in the plant will result in throwing an exception.  So, does the PR work correctly if any kind and number of mimic joints are present?  Or maybe even that is too hard to implement, so we need say "only such-and-such kinds of mimics are allowed" and be sure to have test cases that cover both allowed and disallowed situations.  Maybe the thing you can help with is itemizing the situations where kinds and groupings of mimic constraints end up being distinctly different in practice.  Essentially what I think tcohn is after is expert advice in setting up that test suite.

jwnimmer-tri, 2025-04-02:
(I was also trying not to bother Alejandro and Joe since they are heads-down on critical research, whereas at the moment you (Sherm) have a bit more bandwidth available, while also being an expert on plant topology and coordinates.)

cohnt, 2025-04-04:
1. joint1_index is treated as the independent joint, and joint2_index is treated as the dependent joint, to match [`multibody::internal::CouplerConstraintSpec`](https://github.com/RobotLocomotion/drake/blob/f2ec46196d93672a5280dace0f1c6d719a81f471/multibody/plant/constraint_specs.h#L25).
2. Throwing if there are any non-coupler constraints, if the plant is continuous-time, etc. sounds good to me. But is it possible for someone to add additional coupler constraints that form a "loop"? Is that something I need to be watching for (and writing test cases to verify)?
3. I think the algorithm should work for multiple chains of coupler constraints. (That's certainly something we need to support, e.g., for a bimanual robot with parallel-jaw grippers.)
4. IrisZo is mostly intended for downstream use with kinematic motion planning, so we're not really worried about contact constraints right now.

I do also have some concern about bloating the test cases of IrisZo with a bunch of examples that are ultimately not really related to that algorithm. And @sadraddini has requested having the ability to construct this mapping and use it outside the realm of IrisZo. So this motivates the desire to have a function in MultibodyPlant.

sherm1, 2025-04-04:
I don't think there is anything that would stop someone from forming a loop of coupler constraints -- that would be a special case of redundant (and possibly inconsistent) constraints. Doesn't seem like a good idea though! You could reject systems like that.

jwnimmer-tri, 2025-04-04:
> I do also have some concern about bloating the test cases of IrisZo with a bunch of examples that are ultimately not really related to that algorithm.

Yes.  Having this capability exist outside of IrisZo (e.g., as a helper class or function) seems smart.  Then it can be unit tested by itself, and IrisZo can just simply use it, with only 1-2 smoke tests.

> And sadraddini has requested having the ability to construct this mapping and use it outside the realm of IrisZo

Also a good reason to have it a be a distinct component.

> So this motivates the desire to have a function in MultibodyPlant.

There are tons of components that use MultibodyPlant, but which are not member functions on the plant (see, e.g., rational forward kinematics, trajectory optimization, etc.).  It doesn't need to be a MbP API, nor even in the plant nor tree directory.  It could start out as a component in `planning/iris`.

closed 2026-03-03 by cohnt