RobotLocomotion/drake #22654: Add equals operator for math types

dmcconachie-tri, 2025-02-19:
**Is your feature request related to a problem? Please describe.**

When comparing objects (in C++ or python) it is useful to be able to compare "highly nested" objects with a single `==` test. For example

```python
@dc.dataclass
class Foo:
    bar: RigidTransform = RigidTransform()

@dc.dataclass
class Baz:
    foo: Foo = Foo()

...

class Test(unittest.TestCase):
    def test_something(self):
        a = Baz(...)
        b = Baz(...)
        assert_equal_recursive(self, a, b)
```

where `assert_equal_recursive` does what it says on the box; recursively checks all elements of each object, descending into all (known) container types, producing a nicely formatted error pointing at where in the object structure a check fails. Right now, any tests for equality using `SpatialVelocity` fail unless it is the exact same object rather than the same values as python appears to be falling back to `id(a) == id(b)` due to the lack of an explicit `__eq__` method.

I have spot checked the `SpatialVelocity` and `RigidTransform` C++ classes and I don't see an `operator==` there.

**Describe the solution you'd like**

An `operator==` and `__eq__` defined for math types (base and multibody).

**Describe alternatives you've considered**

1. Special case `assert_equal_recursive` for known/used pydrake math types:
```python
        ...
        elif T == RigidTransform:
            recurse(a.GetAsMatrix4(), b.GetAsMatrix4(), path)
        elif T == SpatialVelocity:
            recurse(a.get_coeffs(), b.get_coeffs(), path)
        ...
```

This is my current workaround.

2. Explicitly/manually use one of the comparison methods above where I first extract the underlying data, and then compare the data using numpy/Eigen. This works when the types are "raw" - i.e.; not nested in some other structure that we are trying to check.

RussTedrake, 2025-02-19:
To be clear, we do have `RigidTransform::IsExactlyEqualTo()`, but haven't mapped it to `operator==`.  I'm not sure why not? 
 @sherm1 -- might you have any history/opinion?

We also have that compartor for `RotationMatrix`.  We don't even have that comparator for `SpatialVelocity`, but I would think it's very easy to add.

@dmcconachie-tri -- the issue title says "for math types".  But there is [other stuff in math](https://drake.mit.edu/doxygen_cxx/namespacedrake_1_1math.html) that i think you don't care about and `SpatialVelocity` is not in math.  Could you help me scope the issue a bit more?

dmcconachie-tri, 2025-02-19:
I was a little vague in the title; you're right. The 2 classes that I *know* of (before writing this comment) that don't have the operators are `RigidTransform` in `pydrake.math` and `SpatialVelocity` in `pydrake.multibody.math`; hence "math".

A quick scan of [drake/multibody/math](https://github.com/RobotLocomotion/drake/tree/master/multibody/math) tells me that all of those classes could use this. They do have a `IsNearlyEqualWithinAbsoluteTolerance()` method.

I'm less familiar with [drake/math](https://github.com/RobotLocomotion/drake/tree/master/math), but it would imagine `Quaternion`, `RigidTransform`, `RollPotchYaw`, and `RotationMatrix` could potentially benefit from an operator.

jwnimmer-tri, 2025-02-19:
BTW Since your goal is a change-detector test and not a functional test, other solutions might be to pickle the `a` and `b` and check that the pickled bytes are identical, or to check that `repr(a)` and `repr(b)` are identical.

The reason we don't have bit-exact equality operators on our float-point types is that because it nearly all cases other than a change-detector test, checking for bit-exact quality on floats is actively wrong.

sherm1, 2025-02-19:
I agree with Jeremy that the reason we didn't overload `operator==()` for fancy types is that it is so likely to be the wrong choice. `IsExactlyEqualTo()` makes it very clear how narrow this comparison is, and leads the way to nice alternatives with a tolerance.

It might be reasonable to overload `==` for SpatialVector since I think the underlying Eigen 6-vector supports that operation anyway. Adding `IsExactlyEqualTo()` would be more clear at the call site though.

jwnimmer-tri, 2025-02-22:
Suggestion: Update `assert_equals_recursive` so that when comparing two objects that don't have an `__eq__`, it checks their `__repr__`.  If the repr string does not contain any `<` then by law it must be a fully-round-trip denotation of the value, and therefore comparing `repr(a) == repr(b)` has the same semantics as `__eq__` so the test can proceed.  If the repr is not suitable, then the comparison would still need to throw an error like it does now.

Does that resolve the problem?

We would prefer not to overload the `__eq__` if we don't have a need.

dmcconachie-tri, 2025-02-22:
Will give it a try when I get a chance.

dmcconachie-tri, 2025-02-22:
Hmm, our `assert_equal_recursive` has a `np_tol` option to make the float comparisons fuzzy which is lost when comparing via `__repr__`. I think our best option may be to simply make `assert_equal_recursive` explicitly aware of pydrake math types as needed for `np_tol` support. There are no so many that it would be a burden to do so.

jwnimmer-tri, 2025-02-24:
One thing we could probably add is a generic protocol for "given an object that's wrapping a numpy with additional semantics, unwrap it to be just the underlying numpy".  In other words, a generic protocol for the calls to `get_vector()`, `matrix()`, `GetAsMatrix34()`, etc. so that the `assert_equal_recursive` could automatically unwrap the numeric-like matrix/vector types.

rpoyner-tri, 2025-02-25:
assigning to component owner @amcastro-tri for further triage.