RobotLocomotion/drake #22664: Add MultibodyPlant constraint to support MuJoCo's tendon/fixed

RussTedrake, 2025-02-24:
As discussed with @joemasterjohn and @amcastro-tri.

MuJoCo implements two types of "tendons". The "[fixed tendon](https://mujoco.readthedocs.io/en/stable/XMLreference.html#tendon-fixed)" is barely a tendon, and is only a modest change away from our current implementation of coupler constraints.  Basically, it implements 
```
length =  offset + ∑ᵢcoef[i]*q[i],
range[0] ≤ length ≤ range[1]
```
where `q[i]` are just multibody our existing multibody positions.  (and we can require them to be simple single-dof joint positions). 

I can follow up with the PR for the parser. I already had a draft [here](https://github.com/RobotLocomotion/drake/commit/e5ed2cb9049f6e26167e940465256f6bb2e6463c) which leveraged the fact that in the special case when `range[0] == range[1]` and there were only two joint specified, then it reduces to a coupler constraint. But I've shelved that PR for now, because it turns out in the case that I care most about `range[0] != range[1]`. This is why i'm requesting the proper implementation.

amcastro-tri, 2025-02-24:
This sounds like a great feature to have in Drake!

Question to you @RussTedrake, so that we can design the API we want to provide. Looking into MuJoCo's docs doesn't give me many answer so I will ask for your specific use case.

To be more specific, is it important that we couple more than 2 q's in the equation above? the coupler constraint in Drake couples only 2 q's. We could, if it solves your problem, to write a tendom model where the tendom has a (maximum) length and we can even add stiffness and damping to medel better a real elastic tendom (users could provide infinite stiffness and/or damping).

In my mind, this allows to model a tendom effectively. If you chain several q's (like pulleys in the real mechanical system) with a cable (tendom) connecting two consecutive pulleys, this constraint would get the job done.

What is confusing about the general equation above is, what physical system is it modeling? do we need that level of generality? would it even get used?  two specific question on this:
 1. What mechanical system does coupling more than 2 q's model?
 2. what does the lower limit in length mean? an upper limit would be the cable's length, but I am not sure I have a good use case for the lower limit?
 3. That is, if we give you 2 q's (two pullyes) with an upper limit (cable's length) and stiffness and damping, does it solve your problem?

RussTedrake, 2025-02-25:
> What physical system it is modeling?

The mujoco docs specifically say: "This element creates an abstract tendon whose length is defined as a linear combination of joint positions."  As a result, I wasn't trying to visualize it as a physical tendon, per se. I think it's modeling power is useful as a generalization of joint limits when the joints are coupled. In this case both lower and upper limits make sense?

> if we give you 2 q's... does it solve your problem?

My current use case has only two joints per "tendon/fixed".  I took a quick look in the mujoco menagerie, and do see several instances of fixed tendons with more than 2 joints (such as in the `hello_robot_stretch3/stretch.xml`).  I think full support of this feature does require > 2 q's. In my mental model of how the implementation would go, I don't think that's any more difficult?

> stiffness and damping

You're right that the mujoco docs due allow specifying stiffness and damping (and springlength, etc). Those should also be easy to include(?), and it would be great. But I don't have an immediate use case for them.

Thanks!

joemasterjohn, 2025-03-18:
As part of validating the implementation of the constraint in #22711, I made a simple sim in MuJoCo and set up the same sim in Drake to compare. The setup is a 1kg box, on a z-axis prismatic joint with gravity. There is a single "tendon" constraint with a coefficient of 1, lower bound of 0 and upper bound of infinity. So effectively `q > 0`, and the constraint acts as joint limits.

As I point out in the docs, formulating the compliant constraint as a linear spring-damper has the well known artifact of producing "action at a distance" when the constraint is violated at `q0`. This is present in the MuJoCo formulation as well.
The area of effect of the action at a distance is proportional to both damping and velocity. So here are 3 simulations of the box being dropped with constraint stiffness of  `k=1e3 N/m` and three levels of damping (no damping, critically damped, over-damped):

## No damping (`d = 0 N⋅s/m`)
|MuJoCo|Drake|
|---------|---------|
|[image] | [image]|

## Critically damped (`d = 30 N⋅s/m`)
|MuJoCo|Drake|
|---------|---------|
|[image] | [image] |

## Over damped (`d = 100 N⋅s/m`)
|MuJoCo|Drake|
|---------|---------|
|[image]|[image]|

You can see the artifact prominently in the (very) overdamped MuJoCo sim. As I also point out in the docs, we get around this in the Drake implementation by only adding the constraints when they get violated, effectively eliminating the artifact. 

I tried pushing the stiffness up as much as I could in MuJoCo and it started going unstable around `k=1e4 N/m`. With the SAP constraint you can crank the stiffness up until it is in the "near-rigid" regime without specifying any damping and it remains stable:

[image]

closed 2025-03-25 by joemasterjohn

jwnimmer-tri, 2025-03-25:
What's the expectation / timeline for adding pydrake bindings for the new public API?

joemasterjohn, 2025-03-25:
> What's the expectation / timeline for adding pydrake bindings for the new public API?

It was an oversight on my part not to include them in the last PR. Added in #22813.