RobotLocomotion/drake #22673: Add initial value to constructor for `Integrator` system

marip8, 2025-02-26:
**Is your feature request related to a problem? Please describe.**

I'm using an `Integrator` system in a diagram that is nested several layers deep inside the top-level simulation diagram, and it needs to have a non-zero initial value (which I know at the time of construction of the system). AFAIK I cannot call `set_integral_value` at the time when I create the `Integrator` system because that method sets the integral value in a context (i.e., ultimately of the full diagram), which does not exist at that time. As a result, I'm forced to "dig" the integrator system out of the diagram after it is constructed to set the initial integral value. Here is some pseudo-code to illustrate the problem:

```python

initial_integral_value = ...
diagram = create_my_robot_controller_diagram(...)  # Creates the diagram in which the `Integrator` is nested
context = diagram.CreateDefaultContext()

integrator = diagram.GetSubsystemByName(...).GetSubsystemByName(...).GetSubsystemByName(...).GetSubsystemByName('integrator')
integrator.set_integral_value(integrator.GetMyMutableContextFromRoot(context), initial_integral_value)
```

There are a number of disadvantages to this approach:
* End users of `create_my_robot_controller_diagram` need to know that the initial value of the integral value needs to be set to some value
* End users of `create_my_robot_controller_diagram` need to know how to get the `Integrator` system out of the top-level diagram

Both of these issues can be addressed in documentation, but that doesn't really help with usability of `create_my_robot_controller_diagram`.

**Describe the solution you'd like**

It would be very beneficial for the `Integrator` system to have an additional constructor that takes the initial integral value so it doesn't have to be set after construction of the top-level diagram.

**Describe alternatives you've considered**

* I tried hacking the initialization unrestricted update event for the integrator (see below), but it seems that the top-level simulation context does not retain these changes.
    
    ```python
    def create_my_robot_controller(...,
                                   initial_integral_value,
                                   ...):
        ...

        integrator = Integrator(size)

        # Create a lambda that can set the initial integral value (known now) whenever the simulation gets initialized
        def on_init(context, _state):
            integrator.set_integral_value(context, initial_integral_value)
        integrator.DeclareInitializationUnrestrictedUpdateEvent(on_init)

        ...

    initial_integral_value = ...
    diagram = create_my_robot_controller_diagram(..., initial_integral_value, ...)
    context = diagram.CreateDefaultContext()

    simulator = Simulator(diagram, context)
    simulator.Initialize()

    # `context` still has original incorrect values for integrator state
    ```

* I also tried sub-classing `Integrator` in Python to have the desired constructor and changing the way the default context is allocated such that it uses the input initial value, but it seems this is not possible due to the following error encountered when that custom class gets added to a diagram builder:
    
    ```
    RuntimeError: Python-extended C++ class does not inherit from pybind11::wrapper<>, and the instance will be sliced. Either avoid this situation, or the type extends pybind11::wrapper<>.
    ```

**Additional context**
N/A

sherm1, 2025-02-27:
Assigned to Jeremy for disposition

jwnimmer-tri, 2025-02-27:
Back to +@sherm1 as [owner for framework](https://drake.mit.edu/issues.html).

sherm1, 2025-02-27:
@marip8 please take a look at PR #22682 -- does that do what you need?

marip8, 2025-02-27:
I think that should be sufficient. Thanks for the quick turnaround in addressing that

sherm1, 2025-03-05:
@marip8 the new Integrator initialization features are now in Drake master if you want to try them out (see PR #22682)

PR #22682 (2025-03-05): [primitives] Allow an initial value to be specified for Integrator
Adds a constructor and setter to the Integrator primitive that allows specification of a non-zero (but numerical) initial value.

Also adds missing tests and cleans up some error handling.

Resolves #22673

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22682)