RobotLocomotion/drake #22676: Python wrapper for refine_mesh

zakharos, 2025-02-27:
Currently `refine_mesh` tool (`//geometry/proximity:refine_mesh`) doesn't have a python interface. Would it be possible to add support for one?

jwnimmer-tri, 2025-02-27:
When filing a feature request, Drake provides a template to help ensure that requests tell a story.  Without the answers to the template, it is difficult for us to put your request into context.

Please explain why [subprocess.run](https://docs.python.org/3/library/subprocess.html#subprocess.run) not good enough?  The input and outputs of the refinement are files on disk, so I don't understand what problem a new interface would solve.  It seems like it would just be 4-line helper function around `subprocess.run`, which should be easy enough to write in the application that needs it?

jwnimmer-tri, 2025-03-28:
Closing for lack of response.  Feel free to discuss on slack or a phone call, if that's easier to provide the context.

nepfaff, 2025-06-01:
A Python interface would be amazing.

**Is your feature request related to a problem? Please describe.**
I want to include this functionality in a Python package. I don't want to make installing Bazel and Drake source an installation step of my package. 

**Describe the solution you'd like**
Python binding for that script so that it can be run after pip installing Drake, similarly to the model visualizer (`python3 -m pydrake.visualization.model_visualizer`). Having a Python function instead of the script would also be okay and slightly preferable.

**Describe alternatives you've considered**
I'm not aware of any.

I'm happy to help with this. I don't think we can bind scripts directly, so my proposed implementation is to factor out a function from [do_main](https://github.com/RobotLocomotion/drake/blob/ce8389362d5b6cfc81707f084bdd67cc9dc55074/geometry/proximity/refine_mesh.cc#L18) and bind that function. Is there anything to consider here when doing this?

PR #23083 (2025-06-17): Add python wrapper for volume mesh refinement
Closes #22676

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/23083)
closed 2025-06-17 by SeanCurtis-TRI