RobotLocomotion/drake #22694: drake-packages.csail.mit.edu (and mirror, etc.) are down

tyler-yankee, 2025-03-03:
### What happened?

The csail.mit.edu download links for Drake are down, but the direct AWS download links are still up and running. While we are working to resolve this, please use drake-packages.s3.us-east-1.amazonaws.com instead of drake-packages.csail.mit.edu.

### Relevant log output

```shell

```

jwnimmer-tri, 2025-03-03:
FYI interestingly enough, it seems the `drake-apt` site is still up.  I wonder what failure mode could hit two of them but not all three.

jwnimmer-tri, 2025-03-03:
https://www.cyberciti.biz/humour/a-haiku-about-dns/

Here's the apt site, with correct DNS:

```
jwnimmer@call-cps:~$ dig drake-apt.csail.mit.edu

; <<>> DiG 9.18.30-0ubuntu0.22.04.2-Ubuntu <<>> drake-apt.csail.mit.edu
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 17125
;; flags: qr rd ra; QUERY: 1, ANSWER: 5, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;drake-apt.csail.mit.edu.	IN	A

;; ANSWER SECTION:
drake-apt.csail.mit.edu. 1800	IN	CNAME	dbl5onf71ufzw.cloudfront.net.
dbl5onf71ufzw.cloudfront.net. 60 IN	A	3.167.99.83
dbl5onf71ufzw.cloudfront.net. 60 IN	A	3.167.99.113
dbl5onf71ufzw.cloudfront.net. 60 IN	A	3.167.99.105
dbl5onf71ufzw.cloudfront.net. 60 IN	A	3.167.99.129
```

Here's the packages site, with broken DNS:

```
jwnimmer@call-cps:~$ dig drake-packages.csail.mit.edu

; <<>> DiG 9.18.30-0ubuntu0.22.04.2-Ubuntu <<>> drake-packages.csail.mit.edu
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 64925
;; flags: qr rd ra; QUERY: 1, ANSWER: 2, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;drake-packages.csail.mit.edu.	IN	A

;; ANSWER SECTION:
drake-packages.csail.mit.edu. 1800 IN	CNAME	sparky.mit.edu.
sparky.mit.edu.		1800	IN	A	18.9.63.12
```

RussTedrake, 2025-03-04:
From @EricCousineau-TRI ... adding `--extra-index-url https://drake-packages.s3.us-east-1.amazonaws.com/whl/nightly/` to pip (instead of drake-packages) results in 
```
WARNING: Skipping page https://drake-packages.s3.us-east-1.amazonaws.com/whl/nightly/drake/ because the GET request got Content-Type: application/x-directory; charset=UTF-8. The only supported Content-Types are application/vnd.pypi.simple.v1+json, application/vnd.pypi.simple.v1+html, and text/html
```
> It looks like how S3 serves the page doesn't allow the right mime type to be used?

jwnimmer-tri, 2025-03-04:
We believe the sites are back up now.  (Users may need to wait for the DNS update to propagate before they see the effect.)

Please let us know if any problems remain.

closed 2025-03-04 by jwnimmer-tri