RobotLocomotion/drake #22700: [ci] Explore using AWS machines for macOS CI builds

jwnimmer-tri, 2025-03-04:
**Is your feature request related to a problem? Please describe.**

We aim to have CI with high uptime, low rate of infrastructure flakes, and low time investment to maintain the infrastructure.  Given a growing number of unsolved recent problems with our current hosting (#22587, #22460, #21971) and the pricing of the brittle solution, it's time to explore alternatives.

**Describe the solution you'd like**

An early experiment with EC2 macOS images shows promise.  A clean build on `mac2.metal` takes around 55 minutes, and `mac2-m2pro.metal` takes around 30 minutes.  For reference, a clean build on current hosting takes around 85 minutes.  Given the pricing, `mac2-m2pro.metal` seems like the best fit.

We should try spinning up a single test job like `aws-mac-arm-sequoia-unprovisioned-clang-bazel-experimental-release` on `mac2-m2pro.metal` as a prototype, to identify what challenges we might face in porting to the new solution.  The idea would be to do the setup by hand while keeping a notebook of what we did, so that in the future if we decide to do a full port we have the notebook to help guide the future automation.

I suggest sequoia for the prototype because that's our forward-looking OS version, and is the one we're commonly struggling with on current hosting.

The test job should NOT use the remote cache.

Note that the instances are priced _per day_ so be sure to set an instance cap of <= 1 during prototyping, so that we never launch two by mistake.

**Describe alternatives you've considered**

We looked at GHA runners as an option.  The free instances are too slow to be good for much, and the paid instances are approximately 10x more expensive (hourly) than our current solution.  Not viable.

If we can't find a good hosting solution, then we could also reduce our macOS support footprint.  We could imagine only running packaging/wheel builds, never any bazel (test) build flavors.  That would probably fit on free GHA.

**Additional context**

N/A

tyler-yankee, 2025-03-10:
After further investigation, I have doubts about the feasibility of this. Happy to discuss more f2f, but documenting my thoughts and sources below.

The main issue we're going to run into is the so-called "scrubbing workflow" that a dedicated host enters ([source](https://aws.amazon.com/blogs/compute/understanding-the-lifecycle-of-amazon-ec2-dedicated-hosts/)):
> In case of an EC2 Mac instance, stopping or terminating a Mac instance initiates a [scrubbing workflow](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/ec2-mac-instances.html#mac-instance-stop) of the underlying Dedicated Host, during which it enters the pending state. This scrubbing workflow includes tasks such as erasing the internal SSD, resetting NVRAM, and more, and it can take up to 50 minutes to complete.

Also, spin up time for an instance isn't terrible, but [isn't great either](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/ec2-mac-instances.html#mac-instance-readiness):
> For an AWS vended AMI with a x86 Mac instance or a Apple silicon Mac instance, the launch time can range from approximately 6 minutes to 20 minutes.

Finally, some notes on billing:
* Billing for the dedicated hosts is per-second for the *host*, regardless of instances ([source](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/dedicated-hosts-billing.html)).
* You aren't charged for the "scrubbing workflow" time, or any time in general that the host is not available ([source](https://aws.amazon.com/blogs/compute/understanding-the-lifecycle-of-amazon-ec2-dedicated-hosts/)).

So, the existing workflow of: spin up instance from image > run job > terminate instance becomes a problem with this wait time on the dedicated host. You can only run one instance per dedicated host at a time ([source](https://repost.aws/questions/QUbLpSsM7pSy2cr_KMdqKxMA/can-we-run-multiple-macos-instances-on-a-single-dedicated-host)). 

We could keep instances running all the time and have Jenkins connect to them, as opposed to spinning them up/down when starting/stopping a job, but the numbers don't add up in that case: we currently have four image configurations (Sequoia/Sonoma x unprovisioned/provisioned), and would only be able to keep two instances up.

tyler-yankee, 2025-03-10:
Perhaps I should reframe the issue; if you add that 50 minutes of spin-down time and ~10 minutes of spin-up time, then end-to-end would roughly be 90 minutes. So not the great improvement that we thought it would be, but still comparable to the current solution and still cheaper if we only run <=2 hosts.

jwnimmer-tri, 2025-03-10:
The progress is good, but to help make an overall decision we should still take some measurements.  It's not clear how often the worst-case delays will happen.

(1) When running multiple unprovisioned job back-to-back on the same dedicated host, in practice what kind of latency throughput can we achieve?

(a) How long is the time from the build becoming the head of the queue to its first Drake setup script inside the image being run?
(b) How much longer until the build result (pass / fail) is made available?
(c) How much longer until the next build starts?

The data for (a+b+c) will give us throughput -- how many builds per day we can run.
The data for (a+b) will give us developer latency -- from the time of a request, how long until the answer.
The data for (a) will give us a sense of warm-up time.

It's possible that a--c will be different for the first build of the day versus being run back-to-back repeatedly.  In the experiment we should run 4+ builds back-to-back on the same dedicated host to see if it improves upon repetition.

For unprovisioned, we care more about throughput than we do latency.

(2) Ditto, but for Provisioned jobs.

For provisioned, we care about both latency and throughput.

jwnimmer-tri, 2025-03-12:
> ... we currently have four image configurations (Sequoia/Sonoma x unprovisioned/provisioned), and would only be able to keep two instances up.

I think it would be plausible to have Sonoma Provisioned as the only flavor in pre-merge and Continuous, by demoting the 1 continuous Sequoia job to Nightly.  Then during working hours, we could keep 1 Sonoma Provisioned instance always running and send the jobs its way for faster response time.

How we feather in any other on-request daytime jobs around that is probably not a big deal.  We could probably send any of the other 3/4'ths of the instance types all onto a another dedicated host that only boots when needed.  Or we could allow up to two smaller-size (cheaper) instances for this instead of the big one.

tyler-yankee, 2025-03-13:
I like both of those ideas; I also realized yesterday that we could keep the same instance running for provisioned jobs which definitely saves on throughput.

I've spent the last couple days tweaking settings on the Jenkins and AWS sides (the spin up/down time makes this frustratingly slow to test), so after today I hope to have some data for unprovisioned jobs. Making a provisioned image would be the next step after that.

tyler-yankee, 2025-03-14:
I was able to run four back-to-back unprovisioned jobs yesterday, and the results are ... not promising, unfortunately. In total, the first started at 7:39 AM and the last finished at 4:51 PM.

> How long is the time from the build becoming the head of the queue to its first Drake setup script inside the image being run?

This data actually isn't technically available, since a job at the head of the queue gets caught behind a scrubbing workflow from a previous run, and therefore mucks with the timestamps you can see in Jenkins. Informally, from what I've seen while working through this, the actual spin up time for an instance is about 5-10 minutes.

> How much longer until the build result (pass / fail) is made available?

On average, 37 minutes.

> How much longer until the next build starts?

On average, 2 hours and change.

So it seems in practice the 50 minutes that Amazon reports for their scrubbing workflow is quite optimistic. To me, running jobs under this workflow -- where we spin down the instance after a single job -- is not a viable throughput.

There is good news: this [blog](https://aws.amazon.com/blogs/compute/quick-restoration-through-replacing-the-root-volumes-of-amazon-ec2/) describes a process whereby you can replace the root volume of a running instance, without having to terminate or stop it. (They actually specifically reference the use case of avoiding the scrubbing workflow on Mac instances.) I'm going to look into this today, but if it works out, we could keep instances running all the time and restore to either the provisioned or unprovisioned AMI at the beginning (or end) of each job.

jwnimmer-tri, 2025-03-17:
From f2f: replacing the root volume seems like it will be pretty brittle (the ssh agent dies), so we should probably not take that route.

tyler-yankee, 2025-03-18:
Provisioned builds run as expected. With the following details: 

* deploying instance from provisioned AMI
* running four builds back-to-back
* no remote cache
* re-using the same instance for all builds (only workspace and temporary directories are deleted in between)

> How long is the time from the build becoming the head of the queue to its first Drake setup script inside the image being run?

Very first job launching an instance gets the same ~10 minute spin up time; other than that, nothing besides the usual checkout steps.

> How much longer until the build result (pass / fail) is made available?

On average, 33 minutes. (Remote cache would speed this up, of course.)

> How much longer until the next build starts?

Essentially immediate.

Still working out the graphics driver errors on the instance, and if that is able to be solved, then we should have a pretty good picture on what this direction can and can't do.

jwnimmer-tri, 2025-03-19:
Okay, here's some progress on the plan ...

Setting aside the "cache server health check" jobs, today we have the following configurations in Production:

(1) Provisioned Production:
- mac-arm-sonoma-clang-bazel-continuous-release
- mac-arm-sonoma-clang-bazel-nightly-everything-release
- mac-arm-sonoma-clang-cmake-nightly-everything-release
- mac-arm-sonoma-clang-cmake-nightly-release
- mac-arm-sequoia-clang-bazel-continuous-release
- mac-arm-sequoia-clang-cmake-nightly-release
- mac-arm-sequoia-clang-bazel-nightly-everything-release
- mac-arm-sequoia-clang-cmake-nightly-everything-release

In short, that's the full product of {sonoma | sequoia} x {bazel | cmake} x {open | everything} with the two most important flavors (both os; bazel only; open only) triggered in Continuous and the rest in Nightly.  Our sole default pre-merge experimental job matches the first one listed.

(2) Unprovisioned Production:

- mac-arm-sonoma-unprovisioned-clang-bazel-nightly-release
- mac-arm-sequoia-unprovisioned-clang-bazel-nightly-release
- mac-arm-sonoma-unprovisioned-clang-cmake-nightly-packaging
- mac-arm-sonoma-unprovisioned-clang-wheel-nightly-release

In short, that's:
- the first two jobs run all open tests again but this time on a freshly-brewed host;
- the last two jobs publish release artifacts for our nightly packages to our registries, and serve as canaries for the monthly release flow; these last two each also have a "staging" twin used as part of the release playbook.

---

So how do we adjust this to better match the constraints of EC2?

(a) Turn off the nightly unprovisioned bazel test jobs (the first two unprovisioned, above). These had more relevance in the past, when we still embraced Homebrew's abuse.  However, now that #21714 is finished, we are less at risk of changes in homebrew breaking us.  Instead, as part of the monthly upgrade ticket, we will bake a new macOS provisioned image every month from the latest homebrew, and use the pre-deployment smoke tests of that new image as the way to flag test failure regressions.

(b) Switch the macOS packaging job (nightly, staging, and experimental) to Provisioned instead of Unprovisioned.  This is actually better, even aside from machine costs.  We don't actually want to ship to users Drake packages which are built from arbitrary, unlocked versions of Homebrew's software.  We bake the macOS image to make a stable starting point, and then we release packages based on that.  Using whatever soup of the day homebrew was shipping at the time a Drake package was made is not easily traceable not reproducible.

Even with just those two changes, we are already in pretty good shape for EC2.  For nearly all builds, we only need exactly two base images, so we can run two bare metal instances, each booted into one of the base images we need (one of each OS flavor), and basically never stop either instance:

Sonoma provisioned jobs:
- mac-arm-sonoma-clang-bazel-continuous-release
- mac-arm-sonoma-clang-bazel-nightly-everything-release
- mac-arm-sonoma-clang-cmake-nightly-everything-release
- mac-arm-sonoma-clang-cmake-nightly-release
- **mac-arm-sonoma-clang-cmake-nightly-packaging** (now provisioned)
- **mac-arm-sonoma-clang-cmake-staging-packaging** (now provisioned)
- ... plus each of their "experimental" twins, including the pre-merge job ...

Sequoia provisioned jobs:
- mac-arm-sequoia-clang-bazel-continuous-release
- mac-arm-sequoia-clang-cmake-nightly-release
- mac-arm-sequoia-clang-bazel-nightly-everything-release
- mac-arm-sequoia-clang-cmake-nightly-everything-release
- ... plus each of their "experimental" twins ...

Deleted jobs:
- ~**mac-arm-sonoma-unprovisioned-clang-bazel-nightly-release**~
- ~**mac-arm-sequoia-unprovisioned-clang-bazel-nightly-release**~

Still TBD:
- mac-arm-sonoma-unprovisioned-clang-wheel-nightly-release

---

To solve the wheel build, we have a few different options:

(a) Rework the macOS wheel builder so that it functions correctly on a provisioned image (i.e., doesn't need to install anything, nor write outside of the build directory, etc.).  This would be a nice benefit for developers, too.  Pwnzing the entire machine during a wheel build is not great in the first place.  Porting the wheel build logic to be bazel rules instead of bash scripts is one way of doing that which we've discussed but never seriously designed nor prototyped yet.

(b) When we need a wheel build, shut down and scrub the Sequoia provisioned instance and boot an Sonoma Unprovisioned instance for a while, instead.  This would probably be fine for the nightly (we could start the wheel one earlier, like 11pm), but the scrub latency would be annoying for daytime experimental wheel builds.

(c) Resurrect the "ship a macOS wheel via GitHub Actions" prototype from last year (originally for the macOS x86 wheels; https://github.com/jwnimmer-tri/drake_github_workflow_prototype).  Use (free) GHA to compile the wheels.  Unlike jobs that need to run tests, the latency to compile one wheel on GHA is not too terrible.

---

So overall, it seems plausible to me to switch to EC2.  Do you agree?  I think the only big risk at this point is whether we can get tests that require GL to run correctly.

tyler-yankee, 2025-03-19:
Regarding the rendering issues -- still working -- I expect it's something with either autologin or some other setup step that I'm missing.

Regarding dropping unprovisioned builds (the second section):
* (a): New images monthly sounds good to me. It just creates a little more overhead for us building images to do it more often, and it might create some delays in the deployment of those new images if we come across unforeseen errors with Homebrew or otherwise. (As you said, these errors are less likely now than they may have been in the past. And the savings in avoiding MacStadium headaches probably makes up for the extra work.)
* I agree with (b).

Regarding the wheel build:
* Perhaps I would need more context on (c) as to what value that would bring -- the tradeoff seems to be saving time and space in the Jenkins queue in exchange for a little more overhead to maintain wheel builds in a separate location. 
* I think the ideal solution is (a), but I don't have enough knowledge/experience there to estimate how much of an effort that would be.
* (b) could be a temporary solution while working on (a), but I don't like it long-term. The latency would be annoying, as you said, and it's inconsistent to change our approach/philosophy for the other unprovisioned builds while running this one as a legacy.

To go back to a point earlier in the discussion, with regular 24/7 uptime of two dedicated hosts running `mac2-m2pro.metal` instances we're still comparable in costs to the existing solution, yeah?

Another thing to keep in mind is how flexible this approach is to future changes in CI job configurations.
* For example, I assume we *eventually* want to run an experimental Sequoia job on PRs?
* Will there be a time where we support three versions of macOS in CI at once (even if just a brief transition period)? We would need a third dedicated host in that case. Perhaps we could downgrade the two less-frequently used ones to the slower instance type, `mac2.metal`, to maintain the same cost.

I have a decent bit of documentation that I've kept throughout this process, so I hope that if we move forward the transition process won't be too painful.

tyler-yankee, 2025-03-19:
Thanks for clarifications.

I was able to get all tests passing, including the rendering ones. There is a step that we do when setting up the MacStadium images to enable autologin, which allows the display server to start when you start the instance. (At first I skipped that step, since we were just prototyping, and then it turns out it *has* to be done via screen share ... anyway, I've documented the relevant steps and pitfalls so we shouldn't have that issue again.)

Assuming we'll continue to move in this direction, I'll keep the AMIs around in AWS (but probably just as a reference, and rebuild new ones next time). Before I shut down the dedicated hosts, is there any further testing needed at this point?

jwnimmer-tri, 2025-03-19:
Good progress!  Okay, as the next step, let's set up a burn-in test.  What might be good?

I'm thinking maybe we try running a `mac-arm-sequoia-clang-bazel-continuous-release`-alike (I suppose with a different job name?), triggered normally by commits to master.  Then we buildcop that and see if it flakes out.  Probably we could run the Sequoia nightlies, too?  Or maybe we run all 4 flavors as Continuous, for an even harder burn-in test?

(This would all still be on 1 bare metal host, not 2.)

mwoehlke-kitware, 2025-03-19:
> Rework the macOS wheel builder so that it functions correctly on a provisioned image (i.e., doesn't need to install anything, nor write outside of the build directory, etc.). This would be a nice [...]

This sounds hard. The wheel builds are fairly reliant on the *install* of Drake being in a well known location. However, it isn't clear what's wrong with the current setup; I do local wheel builds on the same machine that does regular Drake development with no difficulties.

This said, I am told podman works on macOS; have we considered looking into that? It would be nice to have both macOS and Linux wheels using containers to build.

jwnimmer-tri, 2025-03-19:
> However, it isn't clear what's wrong with the current setup; I do local wheel builds on the same machine that does regular Drake development with no difficulties.

For latency reasons, the machine is not rebooted / reimaged between jobs.  So the paths will be "dirty" with artifacts from the prior run.

For the other build flavors, we can simply remove the bazel build tree to start fresh.  The wheel build has a much larger blast radius in terms of what would need cleanup.

> ... have we considered looking into that?

That's also an option worth considering.

tyler-yankee, 2025-03-19:
> ... try running a `mac-arm-sequoia-clang-bazel-continuous-release`-alike (I suppose with a different job name?) ...

I can set up `mac-arm-sequoia-clang-bazel-continuous-release-aws` to run parallel to its non-AWS counterpart.

> Probably we could run the Sequoia nightlies, too?

I don't see why not; I can set up `mac-arm-sequoia-clang-bazel-nightly-release-aws`. More configurations to test isn't a bad thing, and we're paying for the uptime anyway.

> Or maybe we run all 4 flavors as Continuous, for an even harder burn-in test?

Without two dedicated hosts, this would be a pain for latency. I'm in favor of creating a Sonoma AMI and alternating them (at least a few days on each), just to make sure builds with both versions are stable. So I'd create `sonoma` versions of the `sequoia` jobs above and leave one or the other set disabled.

jwnimmer-tri, 2025-03-19:
I was probably unclear.  By 4 flavors, I meant these:
- mac-arm-sequoia-clang-bazel-continuous-release-aws
- mac-arm-sequoia-clang-cmake-continuous-release-aws
- mac-arm-sequoia-clang-bazel-continuous-everything-release-aws
- mac-arm-sequoia-clang-cmake-continuous-everything-release-aws

tyler-yankee, 2025-03-19:
Oh, in that case then sure. Probably worthwhile to also test Sonoma afterwards though, no?

jwnimmer-tri, 2025-03-19:
I am OK not testing Sonoma right away.  If Sequoia works well, then we can plan on moving over to AWS.  If Sonoma doesn't work on AWS, then we drop (or reduce) support for Sonoma.  Our support for the macOS ecosystem is limited by Apple's cloud shenanigans, so if it Apple paints us into a corner with their software license, then we just flow that down to our Apple users to yell at Apple about.  Anyway, hopefully it will work out, or at worst users moving to Sequoia shouldn't be _that_ much trouble.

mwoehlke-kitware, 2025-03-19:
Right, I forgot the issue with containers on macOS is that there are no macOS *containers*. Running Linux on macOS isn't useful.

> The wheel build has a much larger blast radius in terms of what would need cleanup.

Granted, but I *think* that radius is just "the build tree *plus* `/opt/drake*`", which should still be manageable. (Though, worth noting, the wheel builds already nuke that pre-build.)

It seems like the real question is whether it's "safe" to do a wheel build from a fresh checkout on a machine that's been used for prior builds... and I think the answer may already be "yes"?

jwnimmer-tri, 2025-03-19:
> It seems like the real question is whether it's "safe" to do a wheel build from a fresh checkout on a machine that's been used for prior builds... and I think the answer may already be "yes"?

I agree with the question.  Not sure about the answer yet.  Your guess is consistent with the docs ...

https://github.com/RobotLocomotion/drake/blob/b9688191cd58e6794ce292e557bd54e518c7011a/tools/wheel/README.rst#L123-L138

... but we would want to take a little time to cross-check before relying on it.

mwoehlke-kitware, 2025-03-19:
> We would want to take a little time to cross-check before relying on it.

That's reasonable. FWIW, I will point out:

https://github.com/RobotLocomotion/drake/blob/b9688191cd58e6794ce292e557bd54e518c7011a/tools/wheel/macos/build-wheel.sh#L29-L34

https://github.com/RobotLocomotion/drake/blob/b9688191cd58e6794ce292e557bd54e518c7011a/tools/wheel/macos/provision-test-python.sh#L14-L15

mwoehlke-kitware, 2025-03-19:
Also worth noting, when we build more than one wheel, the second build already happens in the same environment that built the first wheel, so if prior builds cause harmful pollution, that second wheel has possibly been polluted this whole time.

As a sort of variation on (b), and preferably in conjunction with feeling confident that wheels can be built on a not-pristine image, we could do nightly — or even just staging — wheel builds on "fresh instances" and use "warm instances" for experimental builds.

tyler-yankee, 2025-03-20:
Alright, I created the four Sequoia continuous jobs referenced above, and ran one myself this morning [here](https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-bazel-continuous-release-aws/3/) to verify the setup. (Note that's using a special branch of `drake-ci`, but only to turn off the remote cache.)

Let's see how it goes!

BetsyMcPhail, 2025-03-20:
I manually launched the remaining jobs - all Continuous jobs must be launched once by hand before they will automatically be run.

jwnimmer-tri, 2025-03-20:
Once we see healthy results in Continuous for a few days, the next step will be to add a trigger for one Experimental flavor to PR jobs, probably `mac-arm-sequoia-clang-bazel-experimental-release-aws`.  (Be sure to announce to developers that the job is a WIP and they are OK to ignore it / not buildcop it.)

tyler-yankee, 2025-03-21:
I've had these jobs disabled since yesterday while resolving a bug related to the Gurobi license - turns out the system location and date/time has to be configured in order to correctly contact the Gurobi token server. The jobs are re-enabled now and test fixed, so we can start monitoring as of today.

jwnimmer-tri, 2025-03-24:
(1) Next steps: re-bake a better image with well-documented / curated fixes, and then switch to it, maybe pair session as docs review.

(2) Add these definitions to D-J-J.

(3) Then after that (maybe soon) also start triggering on PR builds.

tyler-yankee, 2025-03-25:
I know it was already said above, but don't want it to get lost - reminder to update the monthly upgrade ticket once we complete this transition to rebuild a macOS provisioned image.

jwnimmer-tri, 2025-03-26:
> reminder to update the monthly upgrade ticket once we complete this transition to rebuild a macOS provisioned image.

=> #22818

jwnimmer-tri, 2025-03-26:
> > To go back to a point earlier in the discussion, with regular 24/7 uptime of two dedicated hosts running mac2-m2pro.metal instances we're still comparable in costs to the existing solution, yeah?
>
> I still need to double-check the numbers. The only open question is if we do 2x mac2-m2.metal or 2x mac2-m2pro.metal or one of each (using "pro" for the flavor with more jobs, i.e., Sonoma).

For now, we'll start with 2x `mac2-m2pro.metal` for the initial deployment.  After that's stable and the whole transition is finished, we can try switching Sequoia down to `mac2-m2.metal`.  We can afford 2x pro, but it would be nice to shave off some cost if we don't need the throughput of the extra cpus.

BetsyMcPhail, 2025-03-26:
> (4) Add nightly packaging provisioned (don't push to S3, or push with a different path). Experimental can still push.

While discussing the Mac AWS packaging jobs, it didn't fully register with me that they are running on Sonoma. Currently, we're only running Sequoia AWS mac jobs. Is the intention here to spin up a Sonoma host or create Sequoia packaging jobs?

jwnimmer-tri, 2025-03-26:
> Is the intention here to spin up a Sonoma host or create Sequoia packaging jobs?

Ah, I missed that detail also.  In the prototyping stage, I don't mind running them on Sequoia for testing.  I imagine they will work equally (or poorly) well on either OS.

After prototyping, yes ideally we would spin up a Sonoma AMI and have two hosts running, and do the packaging on Sonoma.

That being said, if you'd like to start preparing a Sonoma AMI now and run both machines during the prototype, that's also fine by me.

tyler-yankee, 2025-03-26:
We have two fresh Sequoia AMIs: XCode base, and provisioned. Tomorrow morning, I'll transfer over the Jenkins jobs to use the new image (probably run an experimental job first to test that nothing is terribly wrong).

tyler-yankee, 2025-03-27:
Mac on AWS jobs can successfully upload packages to S3: https://drake-jenkins.csail.mit.edu/view/Mac%20Sequoia%20AWS/job/mac-arm-sequoia-clang-cmake-experimental-packaging-aws/3/.

tyler-yankee, 2025-04-07:
Sonoma nightly packaging looks good with results not uploading to S3: https://drake-jenkins.csail.mit.edu/view/Mac%20AWS/job/mac-arm-sonoma-clang-cmake-nightly-packaging-aws/3/

jwnimmer-tri, 2025-04-07:
From f2f: will turn off Spotlight (to same RAM and probably CPU).

tyler-yankee, 2025-04-10:
Closed as completed, see #22858.