RobotLocomotion/drake #22728: Diagnose problems with libpng 1.6.47 upgrade

jwnimmer-tri, 2025-03-09:
See #22727 where we had to revert a commit from #22670 because it caused console printing.

I don't know if the error messages were true -- possibly some of our test inputs were actually invalid?  It would be helpful to know for sure.

But in any case, even if the png images were invalid, that's still no reason to spam the console.  We either need to suppress the messages entirely (probably best), or else we need to redirect them through a DiagnosticPolicy to hoist them up to the Drake application code that might care.

Aiden2244, 2025-03-17:
The new warnings we are seeing in our test suite are benign and ought to be suppressed. They are emerging because the latest version of libpng, v1.6.47, implemented validation logic for eXIf chunks that was not present in the previous version in order to reflect changes to the PNG standard. These changes expose minor formatting issues in some of the asset images Drake uses to conduct internal rendering tests for VTK. While these formatting issues are true positives, they do not compromise the accuracy of the test results or the overall functionality of the program and are thus targets for suppression.  

That's the long and short of it, but feel free to refer to [this report](https://luxurious-calliandra-0fc.notion.site/Technical-Investigation-Report-libpng-eXIf-Warnings-in-Drake-Tests-1b906c888417804daa88d3e17fb25e07) for a more detailed analysis and explanation.

Aiden2244, 2025-03-21:
@jwnimmer-tri Any thoughts on how to best proceed here? If you feel my reasoning checks out, should we open a new issue to suppress the warnings or address them as a part of this one?

jwnimmer-tri, 2025-03-21:
Yes, it does seem like the warnings are harmless for our use cases, so turning them off sounds good.

PR #22861 (2025-04-07): Suppress benign eXIf warnings from libpng 1.6.47
Closes #22728, unblocks #22862, and towards #22836

Modifies list of custom, drake-specific adjustments to libpng to disable eXIf read support. Disabling this option effectively suppresses the `libpng warning: eXIf: invalid` output that we would observe running the drake vtk rendering tests on Linux. As eXIf chunks are optional and tend to contain image metadata (particularly metadata relevant to images taken with physical cameras, e.g. aperture and shutter speed), reads for these kinds of chunks can be safely disabled for the rendered test imag [... cut]