RobotLocomotion/drake #22748: Reduce Jenkins instance size

williamjallen, 2025-03-12:
The AWS Compute Optimizer indicates that our [Jenkins](https://drake-jenkins.csail.mit.edu) server is currently over-provisioned in every aspect tracked by AWS.  Reducing the instance size was previously believed to be difficult, potentially requiring the entire server to be reprovisioned.  Further investigation suggests that it is much easier than previously thought: simply stopping the instance, changing the instance class, and starting the instance back up is all that's needed.  This issue serves as a place to discuss which instance type is most appropriate moving forward.

jwnimmer-tri, 2025-03-12:
I believe last time we decided that the risk of a performance impact (slow response time) was not worth the $$$ savings.  Maybe you can dig up and xref the prior issue so that we can page it back in.

williamjallen, 2025-03-12:
For reference, the Jenkins server currently runs on `m5n.4xlarge`, which costs roughly $700/month.  Compute Optimizer suggests `r6i.large`, which would cost ~$90/month.  

I found https://github.com/RobotLocomotion/drake/issues/20350 which discusses the EBS volume and https://github.com/RobotLocomotion/drake/issues/19099 which discusses the larger effort to reduce AWS costs, but I'm not aware of any other issues about this topic.

It's worth noting that since the amount of developer time required to change the instance type is almost trivial, we could always switch back to a higher instance class if a smaller instance class proves to be too slow.

jwnimmer-tri, 2025-03-12:
Okay, I was remembering #19099. I guess we didn't write down the thinking over there.  I'm willing to try `r6i.2xlarge` for starters (still half the price, but keeping 64 GiB RAM).

williamjallen, 2025-03-12:
That seems reasonable to me for starters.  If it's fine with you, I can go ahead and give it a try on Friday this week.  If anything ends up being slow for whatever reason, we can roll back to the previous instance class.

jwnimmer-tri, 2025-03-12:
Sounds fine.

williamjallen, 2025-03-21:
Completed today, 03/21/2025.

closed 2025-03-21 by williamjallen