RobotLocomotion/drake #22761: FiniteHorizonLinearQuadraticRegulator no longer works for continuous-time MultibodyPlant

RussTedrake, 2025-03-15:
### What happened?

https://github.com/RobotLocomotion/drake/pull/22609 introduced 
```
    throw std::logic_error(
        "FiniteHorizonLinearQuadraticRegulator only supports system where "
        "either system.IsDifferentialEquationSystem() or "
        "system.IsDifferenceEquationSystem() is true");
```
which seems reasonable, but unfortunately, `IsDifferentialEquationSystem()` is false for `MultibodyPlant` with `time_step = 0.0`, due to:
```
  for (int i = 0; i < num_input_ports(); ++i) {
    if (get_input_port(i).get_data_type() != PortDataType::kVectorValued) {
      return false;
    }
  }
```

This issue appeared for me on the release of v1.39, because my underactuated CI canary fell off its perch.

Do we want to broaden the definition of `IsDifferentialEquationSystem` to take a context and pass if the abstract input ports for `MultibodyPlant` are not connected? 

Clearly we're also missing more local tests that would have caught this.

cc @wei-chen-li 

### Relevant log output

```shell

```

RussTedrake, 2025-03-17:
@sherm1 -- can i get your opinion on this one?

jwnimmer-tri, 2025-03-17:
> Do we want to broaden the definition of IsDifferentialEquationSystem to take a context and pass if the abstract input ports for MultibodyPlant are not connected?

I can't think of a reason why we would want this, or at least not for the case in question.  The case in question already checks that the `input_port_index` being controlled is vector-valued.  Why would a different (fixed) input port being abstract-valued be a problem for the algorithm?  The other input ports are akin to parameters; they are not time-varying.  If we do need a new `IsDifferentialEquationSystem` overload, it doesn't seem to me like this code is a motivating example of the need for it.  Changing the code in question to call `has_continuous_state` seems like *exactly* the right fix.  If there's something more subtle happening, please explain.

wei-chen-li, 2025-03-17:
In my humble opinion, a system governed by state with time derivatives is a differential equation, regardless of whether the input is vector-valued. So perhaps, the checks in [`IsDifferentialEquationSystem()`](https://github.com/RobotLocomotion/drake/blob/9c178ac7df8e12466d7370652c3e64244fb3c9a7/systems/framework/system.cc#L592) could be removed?

It is also worth mentioning that, [`IsDifferenceEquationSystem()`](https://github.com/RobotLocomotion/drake/blob/9c178ac7df8e12466d7370652c3e64244fb3c9a7/systems/framework/system.cc#L567) in its [documentation](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_system.html#aeb5016cfee3ccd384e0a0ba62b4da5bc) requires u to be a vector input, but it does not actually perform the vector-valued check in the implementation.

sherm1, 2025-03-17:
Jeremy's suggestion of avoiding the call altogether seems pragmatic. But it does seem worth rethinking IsDifferentialEquationSystem() if it is too strict. Certainly ports that aren't connected can't cause trouble. But also why is it worried about abstract-valued ports at all? Is it because they aren't differentiable? What would go wrong if we just deleted the port check altogether and made IsDifferentialEquationSystem() equivalent to "has only continuous state variables"?

RussTedrake, 2025-03-18:
The [documentation](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_system.html#a6c3ec92003f2a50d5d2b0c49aa296b86) for `IsDifferentialEquationSystem` states that we have only vector input ports (perhaps `IsVectorDifferentialEquationSystem` would be a more robust name). The intention was to centralize the checks that allow us to think about it as a "simple" differential equation, for control design and analysis.

[#15464](https://github.com/RobotLocomotion/drake/issues/15464) is also relevant.

Unfortunately, MultibodyPlant is anything but simple these days. But it should still be capable of representing simple systems that can be analyzed as such.

PR #22762 (2025-03-24): Avoid rejecting system with floating abstract input port in FiniteHorizonLinearQuadraticRegulator
Resolves #22761

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22762)
closed 2025-03-24 by xuchenhan-tri