RobotLocomotion/drake #22770: Upgrade CDash to v3.9.0

williamjallen, 2025-03-17:
CDash 3.9.0 was [recently](https://github.com/Kitware/CDash/releases/tag/v3.9.0) released.  [drake-cdash.csail.mit.edu](https://drake-cdash.csail.mit.edu) is currently running v3.6.2.  This issue serves as a placeholder for the effort to upgrade the instance to the latest version of CDash.

BetsyMcPhail, 2025-04-17:
The CDash instance was updated to 3.9.0 on 4/11. The next day, CDash errors began, see Slack conversation: https://drakedevelopers.slack.com/archives/C270MN28G/p1744479602644729

More specifically, the CDash instance would be unreachable due to an OOM error. This happened consistently every morning. Eventually it was determined that the coverage jobs (https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-coverage/ and https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-gcc-bazel-nightly-coverage/) were causing the issue.

Also note that recently the CDash instance type was changed to  `t4g.small`.

On 4/16, as a workaround, the CDash instance type was upgraded to `t4g.large`. The next day, both coverage jobs submitted to CDash without errors.

The next step here is to track down the cause of the OOM error.

williamjallen, 2025-04-17:
> Also note that recently the CDash instance type was changed to `t4g.small`

@BetsyMcPhail I'm not aware of the instance size being changed recently, except to bump it to `t4g.medium` and then again to `t4g.large` in response to the OOM issues.  Was that a misunderstanding, or did something change that I wasn't aware of?

BetsyMcPhail, 2025-04-17:
> [@BetsyMcPhail](https://github.com/BetsyMcPhail) I'm not aware of the instance size being changed recently, except to bump it to `t4g.medium` and then again to `t4g.large` in response to the OOM [...]

Sorry, that was my misunderstanding! I must have been confusing it with the recent Jenkins update.

BetsyMcPhail, 2025-05-12:
We did some investigation into this and there was no clear cause. As the instance upgrade has effectively resolved the problem, I'm going to close this issue for now, we can re-open it (or a new issue) if the error reappears.

closed 2025-05-12 by BetsyMcPhail