RobotLocomotion/drake #22790: [ci] Remove mac debug jobs from experimental matrix

jwnimmer-tri, 2025-03-20:
As a follow-up to #21936, in drake-jenkins-jobs, remove this these options:

```
mac-arm-sequoia-clang-bazel-experimental-debug
mac-arm-sequoia-clang-bazel-experimental-everything-debug
mac-arm-sonoma-clang-bazel-experimental-debug
mac-arm-sonoma-clang-bazel-experimental-everything-debug
```

We no longer support Debug builds on macOS because they are so large that they run out of disk space, or generally clog up / harm the cluster and/or cache serve.

BetsyMcPhail, 2025-03-21:
@tyler-yankee please coordinate this with your other in-progress Jenkins updates.

tyler-yankee, 2025-03-26:
Closed as completed via [#164](https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/164).

closed 2025-03-26 by tyler-yankee