RobotLocomotion/drake #22798: Support WITH_USD=ON

xhsonny, 2025-03-22:
I am trying to compile and build drake with `WITH_USD=ON` (Mac m3 chip). It complains about C++20 compatibility. I found this issue https://github.com/RobotLocomotion/drake/issues/21456 and this PR https://github.com/RobotLocomotion/drake/pull/21616. I understand it is a lower priority to add USD support https://github.com/RobotLocomotion/drake/issues/17833. Is there any immediate plan to make `main` work with `WITH_USD=ON`?

 My specific commands without `WITH_USD=ON` below. Whenever I added `WITH_USD=ON` to `bazel build`, it throws an error about C++20 compatibility issue.
```
bazel build -c opt --config=clang --define=WITH_NLOPT=ON //...
bazel run install -c opt --define=WITH_NLOPT=ON -- <BUILD_PATH>
```

Thanks!

jwnimmer-tri, 2025-03-24:
> Is there any immediate plan to make master branch work with WITH_USD=ON?

Not currently, no.  Progress towards the feature is on pause.

As you note, #21456 is the tracking issue for C++20 compatibility challenges, as part of the larger issues #20898 for the USD build system and #17833 for the overall feature.  You can subscribe to #17833 to monitor when the feature is finished.

Closing as duplicate of #21456 etc.

closed 2025-03-24 by jwnimmer-tri

xhsonny, 2025-03-24:
@jwnimmer-tri Appreciate the update. Thanks!