RobotLocomotion/drake #22804: Segmentation Fault in MultibodyPlant::CalcPoseInWorld

cohnt, 2025-03-24:
### What happened?

If you construct a FixedOffsetFrame, don't add it to a plant, and try to compute its pose in the world using a plant_context, you get a segfault. Here's a simple reproduction:
```
import numpy as np
from pydrake.all import (
    AddMultibodyPlantSceneGraph, DiagramBuilder,
    Parser, RigidTransform, RotationMatrix, FixedOffsetFrame)

builder = DiagramBuilder()
plant, scene_graph = AddMultibodyPlantSceneGraph(builder, time_step=0.0)

parser = Parser(plant)
iiwa_file = "package://drake_models/iiwa_description/urdf/iiwa14_no_collision.urdf"
iiwa_model = parser.AddModelsFromUrl(iiwa_file)

# Fix iiwa in the world frame
X_WI = RigidTransform(RotationMatrix.MakeZRotation(np.pi / 4), [0.5, 0, 0])
plant.WeldFrames(plant.world_frame(), plant.GetFrameByName("iiwa_link_0"), X_WI)

plant.Finalize()
diagram = builder.Build()

frame = FixedOffsetFrame("bad_frame", plant.world_frame(), RigidTransform())

diagram_context = diagram.CreateDefaultContext()
plant_context = plant.GetMyContextFromRoot(diagram_context)

frame.CalcPoseInWorld(plant_context)
```

### Version

d1a18eacab

### Relevant log output

```shell

```

mitiguy, 2025-03-26:
@cohnt  Thank you for the bug report.  
I am working to reproduce this in C++.
Was there any exception message that was issued, or was this just a segfault?

jwnimmer-tri, 2025-03-26:
Just a segfault.  The function fails to check the precondition of `get_parent_tree` before calling it:

https://github.com/RobotLocomotion/drake/blob/c946d357be5fd249126cf67dadd5af48567b1657/multibody/tree/frame.h#L221

https://github.com/RobotLocomotion/drake/blob/c946d357be5fd249126cf67dadd5af48567b1657/multibody/tree/multibody_element.h#L108-L111

mitiguy, 2025-03-27:
PR #22821 resolves this issue.

PR #22821 (2025-04-01): [multibody] Fix crash on frame elements not added to a plant
Resolves #22804.

Fixes segmentation fault for a calculation involving a frame that was not added to a MultibodyTree.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22821)
closed 2025-04-01 by jwnimmer-tri