RobotLocomotion/drake #22815: `//examples/allegro_hand/joint_control:py/run_twisting_mug_test` fails non-deterministically on macOS

Aiden2244, 2025-03-26:
### What happened?

The titular test of this issue will occasionally fail on mac builds of drake. See [here](https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-bazel-continuous-release-aws/20/). In my own testing on a local ARM Mac machine, I have observed flaky behavior in this test where it will fail on the first run or sometimes outright fail.

### What operating system are you using?

macOS 15 (Sequoia)

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
[8:49:17 PM]  FAIL: //examples/allegro_hand/joint_control:py/run_twisting_mug_test (Exit 1) (see /Users/ec2-user/workspace/mac-arm-sequoia-clang-bazel-continuous-release-aws/_bazel_ec2-user/5d18ed653821e07bf6c5893c55bee15e/execroot/_main/bazel-out/darwin_arm64-opt/testlogs/examples/allegro_hand/joint_control/py/run_twisting_mug_test/test_attempts/attempt_1.log)
[8:49:18 PM]  [57,036 / 57,426] 8404 / 8732 tests; GenerateMypyStubs bindings/pydrake/pydrake/autodiffutils.pyi; 97s darwin-sandbox ... (10 actions running)
[8:49:18 PM]  FAIL: //examples/allegro_hand/joint_control:py/run_twisting_mug_test (Exit 1) (see /Users/ec2-user/workspace/mac-arm-sequoia-clang-bazel-continuous-release-aws/_bazel_ec2-user/5d18ed653821e07bf6c5893c55bee15e/execroot/_main/bazel-out/darwin_arm64-opt/testlogs/examples/allegro_hand/joint_control/py/run_twisting_mug_test/test_attempts/attempt_2.log)
[8:49:19 PM]  [57,038 / 57,426] 8406 / 8732 tests; GenerateMypyStubs bindings/pydrake/pydrake/autodiffutils.pyi; 98s darwin-sandbox ... (10 actions running)
[8:49:19 PM]  FAIL: //examples/allegro_hand/joint_control:py/run_twisting_mug_test (Exit 1) (see /Users/ec2-user/workspace/mac-arm-sequoia-clang-bazel-continuous-release-aws/_bazel_ec2-user/5d18ed653821e07bf6c5893c55bee15e/execroot/_main/bazel-out/darwin_arm64-opt/testlogs/examples/allegro_hand/joint_control/py/run_twisting_mug_test/test.log)
[8:49:19 PM]  
[8:49:19 PM]  FAILED: //examples/allegro_hand/joint_control:py/run_twisting_mug_test (Summary)
[8:49:19 PM]        /Users/ec2-user/workspace/mac-arm-sequoia-clang-bazel-continuous-release-aws/_bazel_ec2-user/5d18ed653821e07bf6c5893c55bee15e/execroot/_main/bazel-out/darwin_arm64-opt/testlogs/examples/allegro_hand/joint_control/py/run_twisting_mug_test/test.log
[8:49:19 PM]        /Users/ec2-user/workspace/mac-arm-sequoia-clang-bazel-continuous-release-aws/_bazel_ec2-user/5d18ed653821e07bf6c5893c55bee15e/execroot/_main/bazel-out/darwin_arm64-opt/testlogs/examples/allegro_hand/joint_control/py/run_twisting_mug_test/test_attempts/attempt_1.log
[8:49:19 PM]        /Users/ec2-user/workspace/mac-arm-sequoia-clang-bazel-continuous-release-aws/_bazel_ec2-user/5d18ed653821e07bf6c5893c55bee15e/execroot/_main/bazel-out/darwin_arm64-opt/testlogs/examples/allegro_hand/joint_control/py/run_twisting_mug_test/test_attempts/attempt_2.log
[8:49:19 PM]  INFO: From Testing //examples/allegro_hand/joint_control:py/run_twisting_mug_test:
[8:49:19 PM]  ==================== Test output for //examples/allegro_hand/joint_control:py/run_twisting_mug_test:
[8:49:19 PM]  Simple regression test that the twisting_mug demo can perform at least one
[8:49:19 PM]  twist.
[8:49:19 PM]  
[8:49:19 PM]  
[8:49:19 PM]  
[8:49:19 PM]  Running tests...
[8:49:19 PM]  ----------------------------------------------------------------------
[8:49:19 PM]  [2025-03-26 00:49:16.215] [console] [info] Meshcat listening for connections at http://localhost:7000
[8:49:19 PM]  PID frequency [Hz]: 10
[8:49:19 PM]  Kp [Nm/rad]: 13.6208
[8:49:19 PM]  Kd [Nms/rad]: 1.08967
[8:49:19 PM]  .[2025-03-26 00:49:16.365] [console] [info] Meshcat listening for connections at http://localhost:7000
[8:49:19 PM]  PID frequency [Hz]: 10
[8:49:19 PM]  Kp [Nm/rad]: 13.6208
[8:49:19 PM]  Kd [Nms/rad]: 1.08967
[8:49:19 PM]  F
[8:49:19 PM]  ======================================================================
[8:49:19 PM]  FAIL [0.756s]: test_sim_and_control (run_twisting_mug_test.TestRunTwistingMug.test_sim_and_control)
[8:49:19 PM]  ----------------------------------------------------------------------
[8:49:19 PM]  Traceback (most recent call last):
[8:49:19 PM]    File "/Users/ec2-user/workspace/mac-arm-sequoia-clang-bazel-continuous-release-aws/_bazel_ec2-user/5d18ed653821e07bf6c5893c55bee15e/sandbox/darwin-sandbox/23586/execroot/_main/bazel-out/darwin_arm64-opt/bin/examples/allegro_hand/joint_control/py/run_twisting_mug_test.runfiles/_main/examples/allegro_hand/joint_control/test/run_twisting_mug_test.py", line 65, in test_sim_and_control
[8:49:19 PM]      self.assertEqual(sim_process.returncode, None)
[8:49:19 PM]  AssertionError: 0 != None
[8:49:19 PM]  
[8:49:19 PM]  ----------------------------------------------------------------------
[8:49:19 PM]  Ran 2 tests in 5.387s
[8:49:19 PM]  
[8:49:19 PM]  FAILED (failures=1)
[8:49:19 PM]  
[8:49:19 PM]  Generating XML reports...
[8:49:19 PM]  ================================================================================
[8:49:19 PM]  ==================== Test output for //examples/allegro_hand/joint_control:py/run_twisting_mug_test:
[8:49:19 PM]  Simple regression test that the twisting_mug demo can perform at least one
[8:49:19 PM]  twist.
[8:49:19 PM]  
[8:49:19 PM]  
[8:49:19 PM]  
[8:49:19 PM]  Running tests...
[8:49:19 PM]  ----------------------------------------------------------------------
[8:49:19 PM]  [2025-03-26 00:49:17.336] [console] [info] Meshcat listening for connections at http://localhost:7000
[8:49:19 PM]  PID frequency [Hz]: 10
[8:49:19 PM]  Kp [Nm/rad]: 13.6208
[8:49:19 PM]  Kd [Nms/rad]: 1.08967
[8:49:19 PM]  .[2025-03-26 00:49:17.445] [console] [info] Meshcat listening for connections at http://localhost:7000
[8:49:19 PM]  PID frequency [Hz]: 10
[8:49:19 PM]  Kp [Nm/rad]: 13.6208
[8:49:19 PM]  Kd [Nms/rad]: 1.08967
[8:49:19 PM]  F
[8:49:19 PM]  ======================================================================
[8:49:19 PM]  FAIL [0.764s]: test_sim_and_control (run_twisting_mug_test.TestRunTwistingMug.test_sim_and_control)
[8:49:19 PM]  ----------------------------------------------------------------------
[8:49:19 PM]  Traceback (most recent call last):
[8:49:19 PM]    File "/Users/ec2-user/workspace/mac-arm-sequoia-clang-bazel-continuous-release-aws/_bazel_ec2-user/5d18ed653821e07bf6c5893c55bee15e/sandbox/darwin-sandbox/23607/execroot/_main/bazel-out/darwin_arm64-opt/bin/examples/allegro_hand/joint_control/py/run_twisting_mug_test.runfiles/_main/examples/allegro_hand/joint_control/test/run_twisting_mug_test.py", line 65, in test_sim_and_control
[8:49:19 PM]      self.assertEqual(sim_process.returncode, None)
[8:49:19 PM]  AssertionError: 0 != None
[8:49:19 PM]  
[8:49:19 PM]  ----------------------------------------------------------------------
[8:49:19 PM]  Ran 2 tests in 0.872s
[8:49:19 PM]  
[8:49:19 PM]  FAILED (failures=1)
[8:49:19 PM]  
[8:49:19 PM]  Generating XML reports...
[8:49:19 PM]  ================================================================================
[8:49:19 PM]  ==================== Test output for //examples/allegro_hand/joint_control:py/run_twisting_mug_test:
[8:49:19 PM]  Simple regression test that the twisting_mug demo can perform at least one
[8:49:19 PM]  twist.
[8:49:19 PM]  
[8:49:19 PM]  
[8:49:19 PM]  
[8:49:19 PM]  Running tests...
[8:49:19 PM]  ----------------------------------------------------------------------
[8:49:19 PM]  [2025-03-26 00:49:18.405] [console] [info] Meshcat listening for connections at http://localhost:7000
[8:49:19 PM]  PID frequency [Hz]: 10
[8:49:19 PM]  Kp [Nm/rad]: 13.6208
[8:49:19 PM]  Kd [Nms/rad]: 1.08967
[8:49:19 PM]  .[2025-03-26 00:49:18.512] [console] [info] Meshcat listening for connections at http://localhost:7000
[8:49:19 PM]  PID frequency [Hz]: 10
[8:49:19 PM]  Kp [Nm/rad]: 13.6208
[8:49:19 PM]  Kd [Nms/rad]: 1.08967
[8:49:19 PM]  F
[8:49:19 PM]  ======================================================================
[8:49:19 PM]  FAIL [0.754s]: test_sim_and_control (run_twisting_mug_test.TestRunTwistingMug.test_sim_and_control)
[8:49:19 PM]  ----------------------------------------------------------------------
[8:49:19 PM]  Traceback (most recent call last):
[8:49:19 PM]    File "/Users/ec2-user/workspace/mac-arm-sequoia-clang-bazel-continuous-release-aws/_bazel_ec2-user/5d18ed653821e07bf6c5893c55bee15e/sandbox/darwin-sandbox/23610/execroot/_main/bazel-out/darwin_arm64-opt/bin/examples/allegro_hand/joint_control/py/run_twisting_mug_test.runfiles/_main/examples/allegro_hand/joint_control/test/run_twisting_mug_test.py", line 65, in test_sim_and_control
[8:49:19 PM]      self.assertEqual(sim_process.returncode, None)
[8:49:19 PM]  AssertionError: 0 != None
[8:49:19 PM]  
[8:49:19 PM]  ----------------------------------------------------------------------
```

BetsyMcPhail, 2025-03-26:
3/26 https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-bazel-continuous-release-aws/21/

tyler-yankee, 2025-04-02:
4/2: https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-bazel-continuous-release-aws/46/

PR #22859 (2025-04-02): [examples] Disable test case on macOS
Closes #22815. As it's a relatively non-important test, we're just going to disable on macOS -- any *actual* LCM multicast errors would be reported via `//examples/hardware_sim:py/hardware_sim_py_test`.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22859)
closed 2025-04-02 by jwnimmer-tri