RobotLocomotion/drake #22818: [ci] "Refresh Provisioned Images" on the first of each month

jwnimmer-tri, 2025-03-26:
In the "Drake Continuous Integration Details" doc at "Refresh Provisioned Images", we currently specify a refresh every 12 weeks.

Per our new ideas in #22700, instead we should do this once per month, approximately around the 1st of the month.  That aligns it with our other monthly cleanups (externals bump, deprecation removals, etc.) spaced away from our stable releases (which happen mid-month).

Please update the documentation, reminders, etc. to match

Aiden2244, 2025-04-08:
- [ ] Update documentation to reflect a monthly provisioned image refresh
- [ ] Update #buildcop reminder to trigger every month instead of every 12 weeks

Aiden2244, 2025-04-08:
I had Tyler sanity check the documentation and the reminder setup was rather straightforward.

closed 2025-04-08 by Aiden2244