RobotLocomotion/drake #22819: [ci] Jenkins job adjustments for macOS

jwnimmer-tri, 2025-03-26:
As discussed https://github.com/RobotLocomotion/drake/issues/22700#issuecomment-2735043286.

We should make the change now, even running on stadium.

---

Remove these Unprovisioned job definitions:
- mac-arm-sequoia-unprovisioned-clang-bazel-experimental-release
- mac-arm-sequoia-unprovisioned-clang-bazel-nightly-release
- mac-arm-sequoia-unprovisioned-clang-cmake-experimental-everything-release
- mac-arm-sequoia-unprovisioned-clang-cmake-experimental-release
- mac-arm-sonoma-unprovisioned-clang-bazel-experimental-everything-release
- mac-arm-sonoma-unprovisioned-clang-bazel-experimental-release
- mac-arm-sonoma-unprovisioned-clang-bazel-nightly-release
- mac-arm-sonoma-unprovisioned-clang-cmake-experimental-everything-release
- mac-arm-sonoma-unprovisioned-clang-cmake-experimental-release

---

Move these to Provisioned (remove the `-unprovisioned` modifier):
- mac-arm-sonoma-unprovisioned-clang-cmake-nightly-packaging
- mac-arm-sonoma-unprovisioned-clang-cmake-staging-packaging
- mac-arm-sonoma-unprovisioned-clang-cmake-experimental-packaging

Adjust the `drake-ci` logic if necessary so that the remote cache is NOT enabled.  (It's not enabled now, and we don't want to accidentally start enabling it here.)

Also the `drake-ci` logic for package version names (and/or when to push to S3?) might need attention, in case it was conditioned on Unprovisioned.  We want to keep the same upload names and behavior as previously.

Note that this might need to be coordinated with the https://drake.mit.edu/release_playbook.html instructions (to fix some hyperlinks and/or nearby text).

---

There will also changes coming for hweels but we're not ready yet; we'll use a separate ticket for that.

jwnimmer-tri, 2025-03-26:
See also https://github.com/RobotLocomotion/drake/issues/22700#issuecomment-2749017710.  We will also want these aws packaging jobs added:

- mac-arm-sonoma-unprovisioned-clang-cmake-nightly-packaging-aws
- mac-arm-sonoma-unprovisioned-clang-cmake-experimental-packaging-aws
- mac-arm-sonoma-unprovisioned-clang-cmake-staging-packaging-aws (?maybe?)

It you'd prefer to add these at the same time as this ticket, that's fine (just be careful about artifact upload names for 'nightly').

jwnimmer-tri, 2025-03-27:
Yes, thanks.

tyler-yankee, 2025-03-28:
> what's the plan for not uploading Mac AWS nightly packages?

Yes, the branch includes the following logic in `driver/configurations/cmake/step-upload-package-archive.cmake`:
```
elseif(DASHBOARD_JOB_NAME MATCHES "(nightly-packaging-aws|staging-packaging-aws)")
  notice("CTest Status: NOT UPLOADING PACKAGE ARCHIVE BECAUSE MACOS PACKAGING ON AWS IS EXPERIMENTAL")
```
That could be done cleaner long-term (I think we have a `TRACK` variable), but I just wanted to get something working for now since this is just a testing branch anyway.

We only allow that branch for all Mac on AWS jobs currently, so there wouldn't be any case where the packaging jobs don't hit that logic.

closed 2025-04-01 by BetsyMcPhail