RobotLocomotion/drake #22820: [ci] Audit back-to-back macOS AWS Provisioned jobs for cleanliness

jwnimmer-tri, 2025-03-26:
Given that our macOS AWS Provisioned jobs will now be running back-to-back without rebooting the VM each time, we need to be much more careful in our own scripting that one build does not contaminate the next build.

We are already rm'ing the bazel cache ahead of each job, so that's a start.

However, for example in the [mac-arm-sequoia-clang-cmake-continuous-release-aws](https://drake-jenkins.csail.mit.edu/view/Mac%20Sequoia%20AWS/job/mac-arm-sequoia-clang-cmake-continuous-release-aws) logs, we can see the path `/Users/ec2-user/workspace/mac-arm-sequoia-clang-cmake-continuous-release-aws` and child directories of it being used for the build.  I'd guess that Jenkins itself is correctly resetting all of the `git` checkout details to match the new build, but we (or Jenkins) also cleaning up anything else left intact there, like the CMake configure cache?  Or any CMake / Ninja intermediate build outputs?

Does any of the build step modify the source tree?  I know at least for bazel builds, we are probably writing a file into `drake/gen/environment.bazelrc` and I'm not sure if that's being correctly reset (or overwritten) or not?

Is there anything run by `drake-ci` like `brew` something, which changes any system software or settings?  I know we update the network multicast routes (which is fine), but is there anything else?  Any uses of `sudo` perhaps?  I skimmed the logs a bit, but we don't actually echo each command line we're running to the log (*sigh*), so it's not possible to audit from the logs alone.

Please take some time to check / audit here, so we can get ahead of any pollution before turning on pull request triggers.

jwnimmer-tri, 2025-03-26:
+@BetsyMcPhail for delegation.

Aiden2244, 2025-03-31:
After extensive review and local testing, there is minimal risk of build contamination when running concurrent provisioned CI jobs on the same Mac host. In my investigating, I found that the vast majority of all reads and writes were occuring within the WORKSPACE directory, which is cleared between jobs even in provisioned cases. 

Furthermore, I did not find any invocations of `brew` or `sudo` that appeared to be problematic (brew is never even invoked directly during provisioned builds), nor did I observe any modifications to the source tree for provisioned builds. 

The aforementioned `gen` directory is created during the provisioning setup for drake and is not modified or generated by provisioned builds; this should likely not be a problem as the host os and cpu architecture will not be changing between runs.

The only persistent writes that occur during the build process that aren't os related or written to temporary destinations are python cache writes at the `/opt/homebrew/Cellar/python@3.12/...` path (and a few associated symlinks). Between the unprovisioned, clean starting point and the state after the first test had run, `brew` had stored python cache files that persist between runs. While these writes are relatively small and do not effect the integrity of the test results, it might make sense to clear the brew python cache between subsequent builds for extra security.

The files that specifically are written are as follows:

```
Only in /opt/homebrew/Cellar/python@3.12/3.12.9/Frameworks/Python.framework/Versions/3.12/lib/python3.12/__pycache__: trace.cpython-312.pyc
Only in /opt/homebrew/Cellar/python@3.12/3.12.9/Frameworks/Python.framework/Versions/3.12/lib/python3.12/unittest: __pycache__
```

jwnimmer-tri, 2025-03-31:
Excellent, thanks.  Case closed.