RobotLocomotion/drake #22826: GitHub Pull Request Builder is deprecated

tyler-yankee, 2025-03-27:
See https://plugins.jenkins.io/ghprb/. Not only are there security vulnerabilities with the current version, but it is no longer being maintained and has been replaced by [GitHub Branch Source](https://plugins.jenkins.io/github-branch-source/). We should consider migrating to this new plugin, and determine whether or not it replicates the functionality of GHPRB.

tyler-yankee, 2025-04-25:
We may be blocked on this upstream; our code in drake-jenkins-jobs that uses jenkins-job-builder relies on the [trigger-github-pull-request](https://jenkins-job-builder.readthedocs.io/en/latest/triggers.html#triggers.github-pull-request) property, which in turn relies on the GHPRB plugin. 

They seem to have support for [GitHub Branch Source](https://jenkins-job-builder.readthedocs.io/en/latest/project_githuborg.html), but it would be a total refactor of how we define our jobs, likely involving putting Jenkinsfiles in Drake itself.

tyler-yankee, 2025-05-07:
Okay, I've got this to a stable enough state with testing that I'll give a preview/summary moving forward:

* We'll be moving to `Jenkinsfile`(s) stored in Drake, rather than specifying pipelines as DSL in drake-jenkins-jobs. Continuous and nightly jobs will also point to the new `Jenkinsfile`(s), since it's easier to maintain that way, even though otherwise those jobs themselves aren't changing.
* We'll be using multibranch projects for experimental jobs (instead of the current pipeline projects), so the job automatically discovers PRs on Drake (not drake-ci, since there isn't a `Jenkinsfile` there) and gives us the commit to check out accordingly.
* We maintain pretty much all of our existing capabilities re: trigger phrases for experimental jobs (though I haven't tested this bit yet, just going off jenkins-job-builder documentation).
* After trying and failing for a while to roll my own status reporting to GitHub that uses the job name, so that all of the jobs stack up as they do now rather than a single monolith "Jenkins" status check ... I found a [plugin](https://plugins.jenkins.io/github-scm-trait-notification-context/) that does it for us. And it's supported by jenkins-job-builder, so it should work for our current setup.

Eventually, I suppose we could even fold in the continuous and nightly jobs to use this workflow (i.e., including `master` in the multibranch, with a cron for nightly and push triggers for continuous), but that's another step for the future. For now, let's just focus on PRs and moving away from the plugin ... especially considering there might be some CI logic that considers "continuous"/"nightly" in the job name, so it's a little more complicated to change how those are named.

tyler-yankee, 2025-05-14:
@jwnimmer-tri: I think I'm almost ready to put up the Drake side of this for review (the Jenkins jobs still need hacking at), but wanted to ask about external examples. We just added linux-jammy-unprovisioned-external-examples in #22574, and long-term we definitely shouldn't lose that effort, but it's going to take a bit more work to get that job ready than the rest of them. Would you rather I spend more time getting that ready to go all at the same time, or focus on transitioning the existing experimental jobs first and either (a) temporarily lose the external examples job or (b) temporarily hang on to the deprecated plugin only for external examples?

jwnimmer-tri, 2025-05-14:
I am OK losing the DEE niceties.

However, if it's easy just to keep the old plug-in installed and working (e.g., by doing nothing) then we might as well keep it going until we (later) have time to port it over as well?  If the plug-in caused trouble during some Jenkins maintenance upgrade we could just drop it and lose DEE niceties at that time, rather than proactively.

jwnimmer-tri, 2025-07-14:
> Jeremy (and perhaps Betsy): I am looking at some CI/repo permission settings for the new GitHub Branch Source experimental jobs. There is one for the level of trust when discovering PRs from forks that I am unsure on. Which one of these options sounds right to you based on what we currently have?
> Collaborators
> Pull requests from [collaborators](https://developer.github.com/v3/repos/collaborators/) to the origin repository will be treated as trusted, all other pull requests from fork repositories will be treated as untrusted. Note that if credentials used by Jenkins for scanning the repository does not have permission to query the list of collaborators to the origin repository then only the origin account will be treated as trusted - i.e. this will fall back to Nobody. NOTE: all collaborators are trusted, even if they are only members of a team with read permission.From users with Admin or Write permission
> Pull requests forks will be treated as trusted if and only if the fork owner has either Admin or Write permissions on the origin repository. This is the recommended policy. Note that this strategy requires the [Review a user's permission level](https://developer.github.com/v3/repos/collaborators/#review-a-users-permission-level) API, as a result on GitHub Enterprise Server versions before 2.12 this is the same as trusting Nobody.

jwnimmer-tri, 2025-07-14:
Two requirements:
- If a random from the internet opens a PR to do bitcoin mining on our infra, it does NOT run by default.
- If a genuine new contributor opens a PR, we can test it without inviting them as a collaborator.  It's OK to have manual steps like "ok to test".

tyler-yankee, 2025-07-15:
#23194 demonstrates that the support for not building untrusted pull requests is more limited with this plugin than with GHPRB (and IMO they could make their documentation on this more clear).

At it stands, PRs from non-collaborators would build the new jobs, but would checkout the `Jenkinsfile` from a trusted branch (master) rather than their feature branch. But this immediately falls over if someone e.g. creates a script and calls it with `cmake -E` from Drake's CMake.

There is [readTrusted](https://www.jenkins.io/doc/pipeline/steps/workflow-multibranch/#readtrusted-read-trusted-file-from-scm) which can be used to fail the build if the given file has been modified in an untrusted branch. But what, we have to call that for every source file?

jwnimmer-tri, 2025-07-15:
Hmm.  We'll need to find some solution, but I agree that readTrusted isn't any help.

I wonder if the Jenkins overall is where we need to configure this?  Never trigger anything from untrusted users in the first place?

Anyway, please keep poking around to try to figure out how thwart the bitcoin miners somehow.

jwnimmer-tri, 2025-07-16:
> If a genuine new contributor opens a PR, we can test it without inviting them as a collaborator. It's OK to have manual steps like "ok to test".

Let me widen this a little bit -- it's actually OK to require steps on GitHub (like inviting them to some drake-contributors list), so long as a wide selection of people can perform the invitation (e.g., all platform reviewers, TRI dynamics team, and kitware -- i.e., some kind of "approvers" list).  I just wanted to avoid it bottlenecking on having _admins_ like myself or Sherm need to always approve things.  Possibly a new membership group with a dozen people as "maintainers" would be suitable.  We'd need to be sure that adding people to the group doesn't give them any extra powers beyond launching PRs.  (For example, they shouldn't be able to triage / close / re-label etc. any issues.)

tyler-yankee, 2025-07-24:
After some hiccups, the new continuous and nightly jobs should be triggering correctly now. I've compared the job configs in Jenkins to our existing jobs, and saw the continuous jobs trigger on one of the recent merges.

One thing I'm wondering -- when we switch over will we need to run everything manually once? We usually have to do this with continuous jobs regardless, but I'm thinking now for everything because the `Jenkinsfile` is where it pulls parameters, properties, etc. from. If the nightly and continuous triggers are locked in the `Jenkinsfile`, and the job is never kicked off to load that file, it may not ever load the triggers?

tyler-yankee, 2025-07-24:
One more TODO here -- it seems that the `concurrent` property that we used to set in DJJ is now yanked from the `Jenkinsfile` as well. See https://www.jenkins.io/doc/book/pipeline/syntax/#available-options, `disableConcurrentBuilds`. This might complicate matters as we have some continuous jobs that run concurrently and others that don't.

tyler-yankee, 2025-07-30:
Status update: it seems that aside from staging, which has not yet been tested, the new jobs are in great shape.

* After yesterday's changes, the nightly and continuous jobs triggered yesterday / last night as expected.
* The pre-merge job seems to be running on everyone's pull requests.
* The experimental jobs don't run by default but can be run manually or requested with the bot.
* We resolved the issue of pre-merge jobs not running on non-contributor pull requests.

I think we're ready to make the switch, save for staging. I'll be working over the next couple days to prepare the production and experimental/pre-merge pieces separately.

tyler-yankee, 2025-08-05:
Experimental, nightly, and continuous jobs have all been switched over to use the pipelines defined in Drake. The remaining work (aside from the follow-up #23217) is to:

* deploy a "test" staging job and convert it in the same fashion as we have done the others
* uninstall the old plugin from our Jenkins instance

tyler-yankee, 2025-08-08:
With last-minute fixes put in and staging jobs switched over, I'm closing this ticket. We'll have to debug the staging jobs during next week's release, and any follow-up work can be tracked under new tickets.