RobotLocomotion/drake #22848: [ci] Audit / fix back-to-back macOS AWS Provisioned Wheel jobs for cleanliness

jwnimmer-tri, 2025-04-01:
See https://github.com/RobotLocomotion/drake/issues/22820#issue-2950434887 for the background.

As with that ticket, we aim to be able to run wheel builds back-to-back on the same machine, without any chance for prior builds (whether wheel or non-wheel) to pollute the wheel packages that come out.

However, unlike that ticket, we already know that the code is not good enough as of today.  For example, the wheel builder runs `brew bundle` and `brew install` to add some dependencies.  We need to fix that, first; in other words, to work on option (a) from the AWS wheel proposal https://github.com/RobotLocomotion/drake/issues/22700#issuecomment-2735043286:

> (a) Rework the macOS wheel builder so that it functions correctly on a provisioned image (i.e., doesn't need to install anything, nor write outside of the build directory, etc.).

After the `brew` problem is fixed, then next is to check that as claimed in the [README](https://github.com/RobotLocomotion/drake/blob/master/tools/wheel/README.rst) only a few directories in `/opt/foo` are used by the build (and nothing else), and that the wheel builder correctly `rm -rf` those scratch folders beforehand.

+@BetsyMcPhail for delegation.

mwoehlke-kitware, 2025-04-03:
What is going to be responsible for provisioning? (Presumably we don't want the general setup scripts also installing whatever the wheels need?)

jwnimmer-tri, 2025-04-03:
We only have one image, so yes, the general setup scripts should install everything.  Nominally I would say that `setup/install_prereqs --developer` should just install the extra stuff.  After all, we want developers to be able to build wheels themselves locally.  Maybe there is something super scary and we need a new flag?  I didn't see anything scary when I first looked.

Aiden2244, 2025-04-07:
It seems like the problematic `brew` invocations happen exclusively whenever the `_provision()` function defined in `//tools/wheel/wheel_builder/macos.py` is invoked. This function calls `brew bundle` to install the packages listed in `.../image/packages-macos` and then invokes `brew install` directly to install python. 

```python
def _provision(python_targets):
    """
    Prepares wheel build environment.
    """
    packages_path = os.path.join(resource_root, 'image', 'packages-macos')
    command = ['brew', 'bundle', f'--file={packages_path}']
    subprocess.check_call(command)

    for t in python_targets:
        subprocess.check_call(['brew', 'install', f'python@{t.version}'])
```

The `_provision()` function itself is called within the `build()` function of `.../wheel_builder/macos.py`, which itself is dispatched from `.../wheel_builder/common.py`. It appears that the the `provision` option is set to `True` by default, but can be disabled if the user passes in the `--no-provision` command when executing the wheel builder scripts.

```python
    parser.add_argument(
        '--no-provision', dest='provision', action='store_false',
        help='skip host provisioning '
             '(requires already-povisioned host)')
```

The other non-documentation locations where `brew` is mentioned appear to be benign. `.../image/build-wheel.sh` invokes `brew` in order to locate its path for RPATH removal exclusion:

```bash
if [[ "$(uname)" == "Darwin" ]]; then
    HOMEBREW="$(brew config | \grep -E '^HOMEBREW_PREFIX' | cut -c18-)"

    ...

    for lib in "$@"; do
        if [[ "$(uname)" == "Linux" ]]; then
            patchelf --remove-rpath "$lib"
            patchelf --set-rpath "\$ORIGIN/$rpath" "$lib"
        else
            strip_rpath \
                --exclude="$HOMEBREW" \
                "$lib"
            install_name_tool -add_rpath "@loader_path/$rpath" "$lib"
        fi
    done
}
```

`.../macos/provision-test-python.sh` invokes `brew` to find the location of the brew-installed python interpreter

```bash
# NOTE: Xcode ships python3, make sure to use the one from brew.
$(brew --prefix python@$1)/bin/python$1 \
    -m venv /opt/drake-wheel-test/python$1
```

and .../macos/build-wheel.sh` similarly invokes `brew` in order to find the python executable prefix:

```bash
$(brew --prefix python@$1)/bin/python$1 \
    -m venv /opt/drake-wheel-test/python$1
```

To address build contamination concerns, the logic and invocations of `_provision()` ought to be removed from the wheel builder scripts and eventually included as part of the monthly AWS macOS reprovisioning job. If the desired goal is to simply disable the logic without removing it entirely, then implicitly passing the `--no-provision` option when invoking `bazel run //tools/wheel:wheel_builder` or changing the default value for `provision` to `False` is sufficient.

Aiden2244, 2025-04-07:
Also worth noting: the dependencies that the wheel builder scripts need from `brew` that are NOT included in Drake or Drake-ci's dependencies are `bash`, `coreutils`, and any version of `python` that are used to builds specific wheels (e.g. if a 3.10 wheel build is specified but the provisioned image only has 3.12 installed, this script will install 3.10 as well).

mwoehlke-kitware, 2025-04-07:
> the logic and invocations of _provision() ought to be removed from the wheel builder scripts

Agreed, and that should be mostly straight-forward. However, this means that the set of supported Python versions will be duplicated in both `setup/mac` and `tools/wheel/wheel_builder/macos.py` (n.b. `python_targets`). Those locations should get comments noting that the two need to be kept in sync.

mwoehlke-kitware, 2025-04-07:
> the dependencies that the wheel builder scripts need from brew that are NOT included in Drake or Drake-ci's dependencies are `bash`, `coreutils`, and any version of python that are used to build specific wheels

FYI:

- Python(s) are obvious; we build wheels for potentially more versions of Python than we support for other builds.
- I'm pretty sure `coreutils` is so some of the build scripts that get used on both Linux and macOS have consistent implementations of e.g. `cp`. (IIRC, it is `cp` in particular that we want the GNU version.)
- I'm less sure why we need `bash`, but it's probably for the same reason as `coreutils`; so scripts that run on both macOS and Linux have a consistent feature set from `bash`. (Namely, because `/bin/bash` is very old.)

jwnimmer-tri, 2025-04-07:
>  If the desired goal is to simply disable the logic without removing it entirely, then implicitly passing the --no-provision option when invoking bazel run //tools/wheel:wheel_builder or changing the default value for provision to False is sufficient.

This approach probably doesn't help us.

> ... ought to be removed from the wheel builder scripts and eventually included as part of the monthly AWS macOS reprovisioning job. 

Yes, but more precisely we want `setup/install_prereqs --developer` to install the necessary packages, without any change to the footprint of a plain `setup/install_prereqs` that doesn't pass the extra option.

jwnimmer-tri, 2025-04-07:
Also be aware of this note from [1.38.0 release notes](https://drake.mit.edu/release_notes/v1.38.0.html):

> The macOS pip wheels for Python 3.11 are deprecated and will no longer be published after v1.40.0 (approximately 2025-04-15) ([#22581](https://drake.mit.edu/release_notes/_#22581)).

That means that we'll drop 3.11 wheel support very soon, so it doesn't need to be part of `install_prereqs --developer` for this issue.  The only extra version we need for developers is 3.12.

Aiden2244, 2025-04-07:
> Yes, but more precisely we want `setup/install_prereqs --developer` to install the necessary packages, without any change to the footprint of a plain `setup/install_prereqs` that doesn't pass the extra option.

Should this change be included as part of a separate pr?

jwnimmer-tri, 2025-04-07:
If it helps @tyler-yankee with the schedule for image provisioning, then yes it's OK to land it as an ahead-of-time PR.

If it doesn't help with provisioning, then it's probably just as good to have the change as part of the larger wheel-refactor PR.

tyler-yankee, 2025-04-07:
> If it helps @tyler-yankee with the schedule for image provisioning...

We don't have a date pinned down yet, but I'm thinking Wednesday or Thursday for the switch-over. So a PR with the package changes to `setup/install_prereqs --developer` in the next couple days so I can make new images would be much appreciated and make the provisioning process easier! If we hit a hiccup in that timeline, it's not game-breaking.

mwoehlke-kitware, 2025-04-07:
If it works better timing-wise to change (only) `setup` now, I'd be okay adding only 3.12 and *not* yet adding the note to keep `setup` in sync with `wheel_builder`. This way, `wheel_builder` is still the "source of truth" for Drake 1.40. This commit should probably mention that it anticipates removal of provisioning from `wheel_builder`.

What matters IMO is that, when responsibility for installing Python *is* removed from `wheel_builder`, the set of Pythons installed by `setup` should match the set of Pythons that `wheel_builder` builds, and there should be comments noting the need to keep the two in sync.

mwoehlke-kitware, 2025-04-07:
Right now, *binary* distributions need `python@3.12`. For "get something anticipatory to Tyler", I think we can just add `bash` and `coreutils`. When we remove `wheel_builder`'s macOS provisioning, however, I think we'll want to list `python@3.12` in `soucre_distribution/Brewfile` also, even if it's redundant. (The alternative is to list all wheel-supported Python versions *except* the one used for binary distributions, which seems... not ideal.)

jwnimmer-tri, 2025-04-07:
> Right now, binary distributions need python@3.12.

Not on master, they don't (#22827).

mwoehlke-kitware, 2025-04-07:
Bah, I hope I can be excused for being a whole 2-3 *hours* out of date. 🙂 

That does, however, suggest that `soucre_distribution/Brewfile` ought to go ahead and add `bash`, `coreutils`... and `python@3.12` and `python@3.13`, despite the latter being duplicated. Maybe like:

```
# Additional packages needed to build wheels.
brew 'bash'
brew 'coreutils'

# Python(s) for building wheels.  These should be kept in sync with
# `python_targets` in `tools/wheel/wheel_builder/macos.py`.
#
# TODO(...): As of writing, we still support 3.11, however, Drake 1.40 is
# slated to be the last version that supports 3.11, and also the last release
# that does not use AWS. Since the builder is still responsible for installing
# the requisite Python, this is okay. For AWS, however, we intend to remove
# macOS provisioning from the builder, at which point this will become the only
# "source of truth". Meanwhile, we are updating this file in order to provide
# the expected needs for the AWS provisioned images (which we don't expect to
# ever build 3.11 wheels). After 1.40 is released, we should a) drop Python
# 3.11, b) remove macOS provisioning from the wheel builder, and c) remove this
# TODO, in that order.
brew 'python@3.12'
brew 'python@3.13'
```

jwnimmer-tri, 2025-04-10:
> ... it might be a good idea regardless to implement try/catch logic for the cleanup code just to be safe.

I would advise that instead of trying to make the failure handling more robust, rather instead we should simply ensure that at the _start_ of the wheel builder program, the cleanup happens before starting any real work.  (I thought maybe it already did, but anyway we should add it if not.)

> ... `atexit` ... indicating that these files are indeed reset even for unsuccessful runs.

This isn't safe to trust.  The wheel builder program should also prophetically remove the files before getting started, same as the `/opt` stuff.

mwoehlke-kitware, 2025-04-14:
> The wheel builder program should also prophetically (sic) remove these kinds of files before getting started.

(Prophylactically?) The only such file on macOS is the SNOPT tarball, which is either created from a git checkout, or gets copied from "somewhere" to a specific location. In either case, if anything *was* present, it's just going to get overwritten.

More generally, we don't know in advance what `_files_to_remove` will contain. There is no straight-forward way to "pre-nuke" them except to remove existing files before creating them, which is largely superfluous.

> Both the README and test/test-wheel.sh files mention a file named macos/test-wheel.sh. This file does not appear to exist.

It was removed in 3f1fe03c812fce1f4b0. The change to `provision-test-python.sh` seems reasonable.

> I would advise that instead of trying to make the failure handling more robust...

Seconded. If a build breaks, we generally want to keep the pieces in order to be able to do postmortem analysis.

jwnimmer-tri, 2025-04-14:
> (Prophylactically?)

I meant "as a measure of safety, even though sometimes / often it won't matter".

> In either case, if anything was present, it's just going to get overwritten.

I would still like to be better safe than sorry, and have the code delete it actively, near the start of the program.

> There is no straight-forward way to "pre-nuke" them except to remove existing files before creating them, which is largely superfluous.

The straightforward way is to gather all intermittent files into a consolidated temporary directory, with either a fixed name and location or a fixed-pattern name, and then the startup code deletes anything in that temporary arena.

jwnimmer-tri, 2025-04-14:
FYI it's okay to switch to AWS now even with the extra safety, so long as the safety is finished by the next release (1.41, mid-May).

mwoehlke-kitware, 2025-04-14:
> I meant "as a measure of safety, even though sometimes / often it won't matter".

Right; "prophylactically" is what I guessed you meant. The original comment somehow wound up with "prophetically". 🙂

BetsyMcPhail, 2025-04-21:
Wheels are now building on provisioned AWS images, closing.

jwnimmer-tri, 2025-04-21:
Getting the first draft landed was one milestone.  My impression was that we still need to finish #22887 in order to call the fixes from the safety audit complete?

BetsyMcPhail, 2025-04-21:
> Getting the first draft landed was one milestone. My impression was that we still need to finish [#22887](https://github.com/RobotLocomotion/drake/pull/22887) in order to call the fixes from the safety audit complete?

Thanks for catching this, I was overly eager to call it done!

jwnimmer-tri, 2025-05-14:
If you don't know, I don't either, so we'll go with "yes".

closed 2025-05-14 by jwnimmer-tri