RobotLocomotion/drake #22858: [ci] Transition macOS CI from MacStadium to AWS EC2

tyler-yankee, 2025-04-02:
See #22700. Deploying this change will be a delicate process, and require taking down macOS CI for a day to put everything in place.
* Jenkins configuration: 
  - Disable all jobs in `Mac ARM` view.
  - Change MacStadium node labels to `mac-arm-<version>-orka`, so that they aren't used by the existing jobs anymore.
  - Change AWS node labels to `mac-arm-<version>`, so that they are used by the existing jobs.
* drake-jenkins-jobs: 
  - Remove prototyping `-aws` jobs.
  - Set node for all macOS wheel jobs to `mac-arm-<version>-orka`, so that they continue to use MacStadium until there is some resolution from #22848.
  - Remove mac-arm-sonoma-keepalive, mac-arm-sonoma-continuous-cache-server-health-check, and mac-arm-sonoma-nightly-cache-server-health-check jobs. These could be brought back in the future once we establish a cache server for macOS in EC2.
* drake-ci:
  - Remove special logic associated with `-aws` in the job name, and change to `mac`  where applicable.

tyler-yankee, 2025-04-02:
@jwnimmer-tri: We can be flexible to coordinate a day to perform this, but @BetsyMcPhail and I decided it's best to take down all macOS Jenkins jobs for a day as there's lots of moving parts and things that need to get merged at the same time. We don't want to interrupt the next release either.

jwnimmer-tri, 2025-04-02:
That's fine.  I imagine you will be ready for this well before the release date, but we'll sync up as it all gets closer.

tyler-yankee, 2025-04-02:
Leaving as a note for myself: there's a convenient way to disable all Jenkins jobs from a given view from the Script Console:
```
jenkins = Hudson.instance
jenkins.instance.getView("<view>").items*.disabled = true
```

tyler-yankee, 2025-04-08:
Yes, we haven't actually run anything on AWS nightly aside from the Sonoma packaging job. Pulling the plug on Stadium will consist of leaving the current experimental matrix alone, though, and just having the current macOS jobs use AWS instead of Stadium (deleting the `-aws` jobs because they're redundant) (and aside from wheels for now). So the regular nightly/continuous balance will be restored naturally. 

I considered doing that demotion now on the `-aws` jobs, but it's extra maintenance and code review for only another couple days of those special job definitions sticking around.

jwnimmer-tri, 2025-04-08:
Ah, right, it will solve naturally.  That's good enough, I didn't mean to accelerate this tweak, only to make sure it wouldn't be lost.

tyler-yankee, 2025-04-08:
@BetsyMcPhail and @jwnimmer-tri: I am thinking Thursday is the best day to do the switch-over, not only for my own schedule but also to give #22875 time to land so we can re-provision images.

BetsyMcPhail, 2025-04-08:
> [@BetsyMcPhail](https://github.com/BetsyMcPhail) and [@jwnimmer-tri](https://github.com/jwnimmer-tri): I am thinking Thursday is the best day to do the switch-over, not only for my own schedule but [...]

That sounds good to me

jwnimmer-tri, 2025-04-08:
Sure.

tyler-yankee, 2025-04-09:
Sonoma and Sequoia provisioned images are ready to go for tomorrow with the changes from #22875 and #22827.

BetsyMcPhail, 2025-04-14:
Reopening this so we don't forget about updating the buildcop documentation and internal CI docs.

Aiden2244, 2025-04-14:
Buildcop docs should be good to go, pending platform review from TRI (see #22886).

tyler-yankee, 2025-04-15:
One more reminder/question: once wheel builds are ready on AWS and Jenkins jobs are reconfigured so that no jobs are using Stadium anymore, when should we pull the plug on the plugin and all its associated settings?

tyler-yankee, 2025-04-28:
As our MacStadium nodes are no longer accessible, I've deleted the cloud settings from Jenkins and uninstalled the Orka plugin. We have internal docs on how those were configured in case we need to reference anything in the future.

closed 2025-04-28 by tyler-yankee