RobotLocomotion/drake #22869: Exit code 256 on Meshcat tests on macOS

tyler-yankee, 2025-04-07:
On 4/1 buildcops began noticing occasional `MeshcatTest` failures (not necessarily all the same test) coming from `geometry/test/meshcat_test.cc` on macOS:

```
geometry/test/meshcat_test.cc:85: Failure
Expected equality of these values:
  exit_code
  Which is: 256
  0
```

Examples:
* https://drake-jenkins.csail.mit.edu/job/mac-arm-sonoma-clang-bazel-continuous-release-aws/5/
* https://drake-jenkins.csail.mit.edu/job/mac-arm-sonoma-clang-bazel-continuous-release/1213/

rpoyner-tri, 2025-04-08:
Assigning component owner @BetsyMcPhail for further triage.

jwnimmer-tri, 2025-04-11:
I have seen this a couple times.  I think networking on macOS might have uncovered a race condition in the test.

For now, we should mark it as `flaky = True` and hope that retries will solve it.

tyler-yankee, 2025-04-24:
Since the above was merged ~3 days ago, let's wait another week or two, and if no failures come up, call it closed.

jwnimmer-tri, 2025-04-24:
Simpler, I think, is to just close it now.  The buildcop playbook already says to re-open an issue when it reoccurs.

closed 2025-04-24 by jwnimmer-tri