RobotLocomotion/drake #22880: [pydrake] Incorrect return value policy in DiagramBuilder.AddSystem

wei-chen-li, 2025-04-09:
### What happened?

I am not 100% sure about this. In the current pybind11 code, for the function [`DiagramBuilder.AddSystem`](https://github.com/RobotLocomotion/drake/blob/7397b21abdccf13f9717e42be4680c10de7f5de6/bindings/pydrake/systems/framework_py_semantics.cc#L663), the return value policy is (implicitly) `py_rvp::take_ownership`.  However, as per the c++ [documentation](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_diagram_builder.html#a9079ad970bdc0638e0ee7e5efc4d59a4), the returned pointer only references a system owned by the `DiagramBuilder`. Therefore, the return value policy should be `py_rvp::reference_internal`?

I found this problem when debugging another problem mentioned in a [slack thread](https://drakedevelopers.slack.com/archives/C2CK4CWE7/p1741600043753819).

### Relevant log output

```shell

```

rpoyner-tri, 2025-04-09:
I'll take a closer look shortly.

rpoyner-tri, 2025-04-09:
Nevermind. I've confirmed that mere `py_rvp::reference` is sufficient, using your `ekf-binding` branch I pulled a while back.

rpoyner-tri, 2025-04-10:
re (2) some initial code-scanning indicates that most AddToBuilder-like entry points are ok in master. The SharedPointer system binding looked a bit suspicious, and builder.{AddSystem,AddNamedSystem} need fixing.

re(3) no progress yet on test minimization.

wei-chen-li, 2025-04-10:
I have finally came up with a minimal test case that triggers the bug. Filed at #22884. PTAL.

PR #22884 (2025-04-10): [pydrake] Fix DiagramBuilder AddSystem return value policy
Resolves #22880

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22884)
closed 2025-04-10 by rpoyner-tri