RobotLocomotion/drake #22885: [pydrake] Yaml dump may introduce unwanted linebreaks and needs override of default width

naveenoid, 2025-04-10:
Pydrake's [yaml dump methods](https://drake.mit.edu/pydrake/pydrake.common.yaml.html?highlight=yaml_dump#pydrake.common.yaml.yaml_dump) may introduce unwanted line-breaks when yaml dumping long strings. 

The fix would be for the API to accept some form of width  eg. such as pyyaml.dump(yaml_data, filename, width=float("inf")). 

Here is one possible solution (for an inf width use case) and an example unit test verifying intended behavior:
`
def yaml_dump_schema_unlimited(value, filename=None, *, T=None):
    """
    Dump a typed object to YAML using the schema conversion logic,
    but with unlimited string width (i.e., no line wrapping) and
    no schema checking beyond simply converting via to_dict.

    This function mimics the logic of yaml_dump_schema, but it
    passes an extra argument to disable the insertion of line breaks
    in long string values.

    Parameters:
      value: The object to dump.
      filename: (Optional) If provided, write the YAML to this file.
      T: (Optional) The type to use when converting via to_dict.

    Returns:
      The YAML string (if filename is None), or None after writing to file.
    """
    raw = to_dict(value, T=T)
    if filename is not None:
        filename = Path(filename)
        with filename.open("w") as f:
            return pyyaml.dump(
                raw, f, default_flow_style=False, width=float("inf")
            )
    else:
        return pyyaml.dump(
            raw, default_flow_style=False, width=float("inf")
        )
`
Unit test:
`    def test_yaml_schema_unlimited_line_width_with_schema(self):

        @dc.dataclass
        class TestSchema:
            long_field: str

        # Create a long string (e.g., 200 'x' characters)
        long_str = "x" * 200
        # Create an instance of the schema with the long string.
        obj = TestSchema(long_field=long_str)
        # Dump the object using the unlimited-width schema dump.
        yaml_str = mut.yaml_dump_schema_unlimited(obj)
        # Now load it back.
        obj_loaded = mut.yaml_load_schema(TestSchema, data=yaml_str)
        # Verify that the loaded object matches the original.
        self.assert_equal_recursive(obj_loaded, obj)

        # (Optional) Verify that the YAML output itself does not contain unexpected line breaks.
        found = False
        for line in yaml_str.splitlines():
            if line.strip().startswith("long_field:"):
                found = True
                self.assertIn(long_str, line)
                self.assertNotIn("\n", line.split("long_field:", 1)[1])
                break
        if not found:
            self.fail("Field 'long_field' not found in YAML output")`

naveenoid, 2025-04-10:
CC: @ggould-tri

ggould-tri, 2025-04-10:
To be clear, I believe that this is a correctness issue (yaml does not faithfully roundtrip long string values) rather than a feature request?

jwnimmer-tri, 2025-04-10:
The `pydrake.common.yaml.yaml_dump` is a <10 line sugar wrapper around `pyyaml`.  I would be extremely surprised if pyyaml had a round-trip bug.

I believe this is a feature request to output something that humans would enjoy better, in this particular situation.

I offer that if someone wants different yaml formatting, don't even both using pydrake.  Just call `pyyaml` directly.

It is _by design_ that pydrake has *exactly* one canonical output format, so that we have consistency in our output.  If someone wants different, then it's no longer the `pydrake` convention and need not call pydrake at all.

jwnimmer-tri, 2025-04-10:
Ah.  I found the Anzu PR now.  There was some code passing a yaml document as a bash argument without correctly escaping things.  The desire to avoid line wrapping was not a readability problem, rather it's just a band-aid for broken data-marshalling logic.

Closing as "won't fix".

As I said above, if you want `yaml.dump` to use a different `default_flow_style` or `width` than pydrake uses, just call `yaml.dump` directly.  It's very easy, and routing it through `pydrake`'s thin wrapper is to nobody's advantage.

closed 2025-04-10 by jwnimmer-tri