RobotLocomotion/drake #22889: Bump minimum CMake version for Drake installation per deprecation warning

tyler-yankee, 2025-04-11:
Newer versions of CMake (e.g. 3.31.6) report a Deprecation Warning that compatibility with versions of CMake < 3.10 will be removed. ([CMake 4.0](https://cmake.org/cmake/help/latest/release/4.0.html#deprecated-and-removed-features) has already removed compatibility with versions < 3.5.) If possible, we should update the minimum version in `drake-config.cmake.in` for installed copies of Drake, which is currently set at 3.9.

External examples log showing this warning (for `drake-cmake-installed` on Linux Jammy): https://github.com/RobotLocomotion/drake-external-examples/actions/runs/14402627691/job/40397237533

It seems that this was last done in #21804.

jwnimmer-tri, 2025-04-11:
Yes, go ahead anytime.  We only need to support the oldest version in our support matrix, which I believe would be Jammy (CMake 3.22).

jwnimmer-tri, 2025-04-14:
FYI Also related might be the warning alluded to here:
https://github.com/RobotLocomotion/drake-external-examples/issues/381

Possibly the DEE version pins of eigen / fmt / spdlog should be upgraded.

closed 2025-04-22 by BetsyMcPhail