RobotLocomotion/drake #22890: [ci] Update provisioned images to Java 21 for Jenkins

tyler-yankee, 2025-04-11:
Jenkins has issued a warning that they're going EOL on Java 17 in March of 2026. This leaves us plenty of time, but we'll eventually need to update all provisioned images to use Java 21 and sync it up with the version of Jenkins.

The [Java support policy](https://www.jenkins.io/doc/book/platform-information/support-policy-java/) seems out of date (see #22845) and doesn't contain this warning, but we can see it from the admin side.

tyler-yankee, 2025-04-23:
On 2025/04/23, new Ubuntu (Jammy and Noble) unprovisioned images were built with `openjdk-17-jre-headless` uninstalled and `openjdk-21-jre-headless` installed. TBD based on the timing of it all whether May's monthly reprovisioning will use these new images as base, or stick on 17 until we're ready to transition.

tyler-yankee, 2025-04-24:
Internal docs have been updated for macOS and Linux to use the new version. 

I think it actually makes sense to fold this right into the re-provisioning next week. Possibly take CI down for a few hours next Friday and kill three birds with one stone: May's re-provisioning, May's Jenkins updates, and this Java update. (And cross our fingers nothing breaks.)

jwnimmer-tri, 2025-04-24:
BTW also note that https://github.com/RobotLocomotion/drake/pull/22919 is blocked on an AMI update.

tyler-yankee, 2025-05-01:
New unprovisioned and provisioned images are ready to go with Java 21; I'm going to update the version on the Jenkins server itself and do the switch over tomorrow.

tyler-yankee, 2025-05-02:
All images have been converted over, with the exception of the Linux ARM instance for #13514 that I'll be doing later today. Also, the new macOS images can't be tested for another couple hours when the dedicated hosts become available.