RobotLocomotion/drake #22891: Update libbot2 for noble

jwnimmer-tri, 2025-04-11:
See https://github.com/RobotLocomotion/libbot2/tree/drake/scripts/package/linux/ubuntu.

There is a script `jammy/package` that produces two `*.deb` files for Ubuntu 22.04 Jammy.

The request here is to add a new script `noble/package` that does the same thing, but for Ubuntu 24.04 Noble.

In the past, we've uploaded the debs to the Drake Apt site, but we should not do that anymore.  The victory condition here is just merging the PR onto the `drake` branch of `libbot2` with the necessary packaging code.

jwnimmer-tri, 2025-04-11:
Ideally we would have this finished by the end of this month (April).

jwnimmer-tri, 2025-04-14:
Do we care which LCM version we're using?  No, whichever work is okay.

Aiden2244, 2025-04-17:
2 Quick questions:

1. I was able to run the designated program. It didn't crash and it opened XQuartz on my local mac (which I take is a good sign, I have X11 forwarding setup for tri-workstation) but it did not display anything. Furthermore, I got these errors:
```
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
```
does this indicate a successful run, or is the build broken?

2. To what extent do we care about this library being backwards compatible with older versions of ubuntu? While the script is working for noble, all other versions are broken due to the package `freeglut3` transitioning to `libglut3.12.` The latter package is not available on ubuntu releases prior to noble AFAIK. I added some conditional cmake logic to `cpack.cmake` to account for the fact that noble has slightly different dependencies from all prior versions of ubuntu, but even with jammy targeting the `freeglut3` package as a dependency (which it was doing before I made any changes to the repo), I was getting the following error:

```
/opt/libbot2/0.0.1.20250417/bin/bot-wavefront-viewer:
 Position Independent Executable: yes
 Stack protected: yes
 Fortify Source functions: yes
 Read-only relocations: yes
 Immediate binding: yes
 Stack clash protection: unknown, no -fstack-clash-protection instructions found
 Control flow integrity: yes
++ command -v bot-wavefront-viewer
+ missing_libraries /opt/libbot2/0.0.1.20250417/bin/bot-wavefront-viewer
++ ldd /opt/libbot2/0.0.1.20250417/bin/bot-wavefront-viewer
++ grep -c found
+ [[ 2 != 0 ]]
+ echo 'Missing libraries for /opt/libbot2/0.0.1.20250417/bin/bot-wavefront-viewer:'
Missing libraries for /opt/libbot2/0.0.1.20250417/bin/bot-wavefront-viewer:
+ ldd /opt/libbot2/0.0.1.20250417/bin/bot-wavefront-viewer
+ grep found
	libglut.so.3 => not found
	libglut.so.3 => not found
+ exit 1
```

In any event, if these scripts truly are broken and some additional work is required to get them functioning for jammy, I would imagine that those changes would be introduced as a separate pr anyway, but I felt it was worth mentioning here.

jwnimmer-tri, 2025-04-17:
> does this indicate a successful run, or is the build broken?

I don't know.  Forwarding Noble X server over to macOS X client is not a supported workflow.  The question is whether it works for a logged-in user on Noble, directly.  Someone will need to test on an actual full-blown Noble desktop login.

IIRC Noble's user login screen offers an option to choose Wayland or X?  It's OK if the program only works in non-Wayland login mode.

> To what extent do we care about this library being backwards compatible with older versions of ubuntu?

Parsing this all out from pages of prose is difficult.  Just open a PR (or link to a patch) with the proposed edits, and I can report back if there are any problems once I can see the _code_.

jwnimmer-tri, 2025-04-19:
Okay, I went and found the branch and opened https://github.com/RobotLocomotion/libbot2/pull/24 so that I could look at the code to offer guidance.

jwnimmer-tri, 2025-04-19:
> To what extent do we care about this library being backwards compatible with older versions of ubuntu?

None at all.  TRI builds packages for Jammy and earlier using a pinned git sha, so nothing we do in this pull request will break them.

Aiden2244, 2025-04-21:
See https://github.com/RobotLocomotion/libbot2/pull/25

jwnimmer-tri, 2025-04-24:
As best we can tell so far, the new packages are working OK internally.

closed 2025-04-24 by jwnimmer-tri