RobotLocomotion/drake #22896: [tutorials] LaTeX salad in debug_mathematical_program

jwnimmer-tri, 2025-04-14:
In `tutorials/debug_mathematical_program.ipynb` on Deepnote (see [link](https://deepnote.com/workspace/Drake-0b3b2c53-a7ad-441b-80f8-bf8350752305/project/Tutorials-2b4fc509-aef2-417d-a40d-6071dfed9199/notebook/debugmathematicalprogram-febc86880e0b4ad8a92c281abce9f463)) the cell `display(Markdown(prog.ToLatex()))` does not render correctly anymore..

[image]

Seen during v1.40.0 deployment, but might have happened earlier and we didn't notice.

jwnimmer-tri, 2025-04-14:
Ditto for the same code in `tutorials/mathematical_program.ipynb`.

jwnimmer-tri, 2025-04-14:
FYI @RussTedrake @hongkai-dai

RussTedrake, 2025-04-20:
it looks like it's the \begin{align} that isn't working any more.
it seems likely that deepnote has broken it broadly?

https://deepnote.com/app/sebastian-kvalden/Assignment-5-09170168-e0bd-46d3-b03c-30e497a433ef