RobotLocomotion/drake #22910: MultibodyPlant can report if vs are the same as qdots

SeanCurtis-TRI, 2025-04-16:
**Problem statement**

There are applications that depend on whether or not the iᵗʰ velocity is the time derivative of the iᵗʰ position (i.e., `vᵢ = ẋᵢ `). This condition is trivially true for many of our joint types, but not all.

This will *not* be true for any joint which incorporates a quaternion in its `q`. The `q` will have four dofs but the `v` will only have three (ignoring any translational component). This has given rise to the practice of detecting if a piece of state satisfies the `vᵢ = ẋᵢ ` by simply comparing the *number* of positions and the *number* of velocities. Historically, this has been a reasonable proxy.

However, that has *not* been true since the introduction of `BallRpyJoint`. In this case, the number of `q`s matches the number of `v`s, but `vᵢ = ẋᵢ ` is _not_ satisfied. So, we need an alternate solution for applications that truly care.

**Proposal**

Mobilizers know if they satisfy `vᵢ = ẋᵢ `. They have a virtual method: `Mobilizer::is_velocity_equal_to_qdot()`. However, `Mobilizer` in `internal`.  But we can leverage that functionality in the public API.

We'll introduce the following methods on `MultibodyPlant`:
  - `bool AllVelocitiesAreQDots() const`
  - `bool AllModelVelocitiesAreQDots(ModelInstanceIndex) const`
  - `bool AllModelVelocitiesAreQDots(const std::string&) const`
No `Context`-based APIs are necessary as this is a property of the `plant` and not its state.

I'd actually advocate something a bit more aggressive. I'd recommend that these quantities be computed during `Finalize()` and stored in the `MultibodyPlant` so that the questions can be answered very quickly. Not all applications can (or should) memoize the answer to the question based on the plant, but may depend heavily on the answer being true over and over (as a runtime validation step). So, memoizing the answer in `MultibodyPlant` would allow downstream applications to find out fast enough to freely validate.

As long as we're here, we'd also expose add the method `bool Joint::AllVelocitiesAreQDots() const` (the plant would make use of this API to answer its own query).

**Alternatives**

1. Downstream applications can create explicit list of known joint types that are compatible with the `vᵢ = ẋᵢ ` requirement and examine all joints to see if the adhere to the list.
2. If we *only* had the `Joint` method proposed above, it could iterate through all queries and rely on the `Joint` to answer the question robustly.

Neither of these are necessarily performant and attempts to memoize the result *outside* the plant is fraught, at best.

SeanCurtis-TRI, 2025-04-16:
cc @sherm1 @amcastro-tri @jwnimmer-tri

jwnimmer-tri, 2025-04-16:
We have `bool MbP::IsVelocityEqualToQDot() const` already.  Is the only thing missing adding a per-model-instance overload to that?

SeanCurtis-TRI, 2025-04-16:
What can I say but, "Duh!"

I agree, for `DofSet` we only need plant-wide answers.  However, we *do* need it to be faster than iterating over all mobilizers every time we ask. But as the current implementation already has a todo suggesting the value be cached, then we could simply cache the value at finalization and `DofSet` would be content.

(Although, the SnakeCase spelling of the current method belies the performance we'd be backing it up with and hoping to get in `DofSet` -- so perhaps a new method and a deprecation of the old.)

Also, it's documentation is incomplete; it fails to mention (or test) that it can only be called post finalize.

sherm1, 2025-04-16:
FWIW I think it would be fine to leave the CamelCase spelling as it is but document that it caches the result at Finalize() and is thus very fast. We're not obligated to use snake_case for fast functions.

jwnimmer-tri, 2025-04-16:
Re-reading the OP, I think the only fundamentally missing piece is `bool Joint::is_velocity_equal_to_qdot() const`.  All of the stuff about grouping by-model and by-name is just sugar.  If we had a joint getter to report the prime fact, then users could easily write their own sugar.  The central problem is just that one method being hidden on Mobilizer instead of pushing it out to the Joint.