RobotLocomotion/drake #22926: IPOPT Solver Options Not Promoting Int to Float

cohnt, 2025-04-24:
### What happened?

If you set the max wall clock time in IPOPT to an integer value, it throws an error, but it works correctly if it's a double. Maybe connected to #18024, but it seems like the error is specifically in the IPOPT interface.

```
from pydrake.all import MathematicalProgram, IpoptSolver, SolverOptions

prog = MathematicalProgram()
x = prog.NewContinuousVariables(1)

options = SolverOptions()
# options.SetOption(IpoptSolver().solver_id(), "max_wall_time", 10) # Fails
options.SetOption(IpoptSolver().solver_id(), "max_wall_time", 10.0) # Works

result = IpoptSolver().Solve(prog, solver_options=options)
```
Bug output:
```
Tried to set Option: max_wall_time. It is a valid option, but it is of type  Number, not of type Integer. Please check the documentation for options.

### max_wall_time (Real Number)  ###
Category: Termination
Description: Maximum number of walltime clock seconds.
0 < (1e+20) <= +inf
Traceback (most recent call last):
  File "/home/tommy/temp.py", line 10, in <module>
    result = IpoptSolver().Solve(prog, solver_options=options)
RuntimeError: Error setting IPOPT integer option max_wall_time=10
```

### Relevant log output

```shell

```

jwnimmer-tri, 2025-04-24:
To solve this, I believe we would need to ask the `IpoptApplication` for its `RegOptions()` and then check the registered type of the option name, to know if we need to promote `int` to `float`.  This would be in `solvers/ipopt_solver.cc` inside our `SetAppOptions`.

sherm1, 2025-04-25:
Assigned to @hongkai-dai for disposition.

PR #22927 (2025-04-29): [solvers] IpoptSolver accepts integers for floating-point options
Resolves #22926

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22927)