RobotLocomotion/drake #22944: "GET returned 400 Bad Request" on Github downloads

xuchenhan-tri, 2025-04-28:
Github downloads failed with the following example error message

`ERROR: no such package '@@zlib+//': java.io.IOException: Error downloading [https://github.com/madler/zlib/releases/download/v1.3.1/zlib-1.3.1.tar.gz] to /media/ephemeral0/ubuntu/workspace/linux-jammy-unprovisioned-clang-bazel-experimental-everything-debug/_bazel_ubuntu/ae4452c4728d124f46a0f630d0202673/external/zlib+/temp5898176615685476697/zlib-1.3.1.tar.gz: GET returned 400 Bad Request`

This has happened more than once recently at 

https://drake-jenkins.csail.mit.edu/job/linux-jammy-unprovisioned-clang-bazel-experimental-everything-debug/31/

and 

https://drake-jenkins.csail.mit.edu/view/Production/job/linux-noble-gcc-bazel-continuous-release/927/

BetsyMcPhail, 2025-04-29:
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-gcc-bazel-continuous-everything-release/1228
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-clang-bazel-continuous-everything-release/1649

BetsyMcPhail, 2025-04-30:
4/29
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-gcc-bazel-continuous-everything-release/1233 https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-noble-clang-bazel-continuous-release/934/

BetsyMcPhail, 2025-04-30:
4/30 https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-continuous-everything-release/1234/

TimotheusBachinger, 2025-04-30:
Hi drake-guys,

we're also heavily facing exactly the same issue in combination with bazel mod + github:
```
Error in download: java.io.IOException: Error downloading [https://github.com/bufbuild/buf/releases/download/v1.27.0/buf-Linux-x86_64] to /home/jenkins/.cache/bazel/_bazel_jenkins/511ee6f5c2c78f955f5cedd45e175c7b/external/rules_buf~~ext~rules_buf_toolchains/buf: GET returned 618 jwt:jwt-not-provided 
```

It all started on 28th of april 6pm CEST. 
Currently we don't see any obvious changes in our repo nor infrastructure so we assume it is an issue on github.

Do you already have a clue?

jwnimmer-tri, 2025-04-30:
Thanks for asking!  Do you have your own issue posted anywhere else, where we could subscribe to follow in case you solve it?

Sorry, we don't really any insight yet.  Like you, I also assume it's either that GitHub is broken, or possibly (in our case, at lest) some AWS EC2 proxy is broken, since we've only ever seen this in EC2, not locally.

Traditionally (when in WORKSPCAE mode), Drake has self-hosted backup mirrors for all externals, but we stopped doing that for externals that come from BCR.  What I was going to do towards this ticket was try to figure out how to add fallback-mirror URLs to the BCR items, so that in case there is a download flake from `github.com` that Bazel would fall back to the Drake mirror.  (I'm not sure if that will be possible without major surgery like hosting our own registry, but I guess I'll find out.)

TimotheusBachinger, 2025-04-30:
Hey  @jwnimmer-tri !
> Do you have your own issue posted anywhere else, where we could subscribe to follow in case you solve it?
not yet - it is astonishingly difficult to report such issue torwards github - I was unable to find the right spot for it....

> I was going to do towards this ticket was try to figure out how to add fallback-mirror
I am not sure if this helps you, but today we tried to do this in such way:
https://github.com/Checkmk/checkmk/commit/5dff11b07a8d96ccd439d4b1b8f4d8af2b58d987

BetsyMcPhail, 2025-04-30:
https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-leak-sanitizer/1557
https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-continuous-everything-address-sanitizer/1560
https://drake-jenkins.csail.mit.edu/job/linux-jammy-unprovisioned-gcc-bazel-continuous-mirror-to-s3/1638

jwnimmer-tri, 2025-05-01:
To help me follow along, I put together of the error messages scraped from all of the failures so far:

```
Error downloading [https://github.com/bazelbuild/rules_rust/releases/download/0.56.0/rules_rust-0.56.0.tar.gz]: GET returned 618 jwt:jwt-not-provided
Error downloading [https://github.com/bazel-contrib/bazel_features/releases/download/v1.22.0/bazel_features-v1.22.0.tar.gz]: GET returned 400 Bad Request
Error downloading [https://github.com/bazelbuild/rules_rust/releases/download/0.56.0/rules_rust-0.56.0.tar.gz]: GET returned 618 jwt:jwt-not-provided
Error downloading [https://github.com/bazelbuild/rules_java/releases/download/8.11.0/rules_java-8.11.0.tar.gz]: GET returned 618 jwt:jwt-not-provided
Error downloading [https://github.com/bazelbuild/rules_shell/releases/download/v0.3.0/rules_shell-v0.3.0.tar.gz]: GET returned 400 Bad Request
Error downloading [https://github.com/bazelbuild/rules_shell/releases/download/v0.3.0/rules_shell-v0.3.0.tar.gz]: GET returned 400 Bad Request
Error downloading [https://github.com/bazelbuild/rules_rust/releases/download/0.56.0/rules_rust-0.56.0.tar.gz]: GET returned 618 jwt:jwt-not-provided
Error downloading [https://github.com/madler/zlib/releases/download/v1.3.1/zlib-1.3.1.tar.gz]: GET returned 400 Bad Request
Error downloading [https://github.com/bazel-contrib/bazel_features/releases/download/v1.22.0/bazel_features-v1.22.0.tar.gz]: GET returned 400 Bad Request
Error downloading [https://github.com/bazelbuild/apple_support/releases/download/1.17.1/apple_support.1.17.1.tar.gz]: GET returned 618 jwt:jwt-not-provided
```

tyler-yankee, 2025-05-15:
5/14 https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-bazel-continuous-release/513/

tyler-yankee, 2025-05-19:
5/18 https://drake-jenkins.csail.mit.edu/job/linux-jammy-unprovisioned-gcc-bazel-nightly-everything-release/1051/

This one was not one of the error codes already mentioned above:
```
ERROR: @@spdlog+//:header_only :: Error loading option @@spdlog+//:header_only: no such package '@@rules_license+//rules': java.io.IOException: Error downloading [https://github.com/bazelbuild/rules_license/releases/download/1.0.0/rules_license-1.0.0.tar.gz] to /home/ubuntu/.cache/bazel/_bazel_ubuntu/3a7067e5fab035a829c6a26cf884bc28/external/rules_license+/temp13697411728103932089/rules_license-1.0.0.tar.gz: GET returned 500 Internal Server Error
```

tyler-yankee, 2025-06-16:
6/16 https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-cmake-nightly-everything-debug/485/