RobotLocomotion/drake #22973: Capsules parsed from MJCF have the wrong length

vincekurtz, 2025-05-05:
### What happened?

For a simple MJCF model like this double pendulum:
```xml
<?xml version="1.0"?>
<mujoco model="robot">
  <worldbody>
    <body>
      <joint type="hinge" axis="0 1 0" pos="0 0 0.1" damping="1e-5"/>
      <geom type="capsule" size="0.01 0.1"/>
      <body>
        <joint type="hinge" axis="0 1 0" pos="0 0 -0.1" damping="1e-5"/>
        <geom type="capsule" size="0.01 0.1" pos="0 0 -0.2"/>
      </body>
    </body>
  </worldbody>
</mujoco>
```
the correct model (parsed by mujoco) should look something like this
[image]
but in Drake (`bazel run //tools:model_visualizer -- double_pendulum.xml`) it looks like this:
[image]
Changing the capsule length (second field of `size`) appears to have no effect. The capsule radius (first field of `size`) does seem to be parsed correctly.

This was not a problem previously: I noticed this after merging `master` into a development branch. I'll bisect on the git history to try to find the source. 

### Version

08d2d75781

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell

```

jwnimmer-tri, 2025-05-05:
Yup.  #22824 threw away a `Parse...` call for the scale which was inside an `if`.  That's why `if` conditions shouldn't have side-effects!

vincekurtz, 2025-05-05:
As far as I can tell everything parses correctly up through SHA `6f150b2663`, indicating that the problem is (the more recent) #22932

jwnimmer-tri, 2025-05-05:
Oops.  Indeed, #22824 didn't throw away the `Parse...`, the diff just led me astray.

vincekurtz, 2025-05-05:
Going back to `master` and reverting #22932 manually does fix the issue. Should I open a PR with that hotfix? It looks there are some [upstream changes](https://github.com/mrdoob/three.js/releases) to the capsule geometry in three.js that could be the root cause.

jwnimmer-tri, 2025-05-05:
Yes, we either need to revert #22932 or fix-forward with revised code that works correctly.

jwnimmer-tri, 2025-05-05:
If the only relevant upstream change was the length vs height factor of 2, maybe it's easy to just fix our `meshcat.cc` reifier?

vincekurtz, 2025-05-05:
As far as I can tell the height is not getting parsed at all and is just fixed at 1 meter, rather than being off by a factor of 2

jwnimmer-tri, 2025-05-05:
Our message probably needs the rename:
https://github.com/RobotLocomotion/drake/blob/08d2d75781cb0b7d44ff2da5860b4408f712ed5e/geometry/meshcat_types_internal.h#L227

vincekurtz, 2025-05-05:
Thanks for the pointer! Looks like the fix is as easy as changing `length` to `height` - #22975 should do the trick

rpoyner-tri, 2025-05-06:
pro forma assignment of issue to @vincekurtz per platform checklist.

vincekurtz, 2025-05-15:
Resolved by #22975

closed 2025-05-15 by vincekurtz