RobotLocomotion/drake #22978: `//planning/iris:iris_zo_test` times out in CI

BetsyMcPhail, 2025-05-06:
### What happened?

4/16: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-everything-valgrind-memcheck/534/

5/6: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-clang-bazel-nightly-everything-valgrind-memcheck/555/

### What operating system are you using?

Ubuntu 22.04

### Relevant log output

```shell

```

rpoyner-tri, 2025-05-06:
I'll take a first look. Could be that the memcheck test timeouts should just be cranked up in a bazel.rc somewhere.

rpoyner-tri, 2025-05-06:
Times for this test under these conditions vary independently of  code changes (from 20ish minutes to hours); I'm somewhat stumped.  Maybe we exempt this test from some builds, or reduce the amount of work it does under some builds? 🤷 

FYI @cohnt

jwnimmer-tri, 2025-05-06:
See #22149).  We are supposed to be skipping memcheck builds on tests that call commercial solvers.  If this test ends up calling a commercial solver, it shouldn't be running under memcheck.

jwnimmer-tri, 2025-05-06:
Sorry, let me clarify: I mean binary-only solvers, not commercial solvers.  Running with SNOPT is OK.  Running with Gurobi or Mosek is not.

cohnt, 2025-05-07:
Some extra clarification -- IrisZo needs to solve QPs when we set a certain option, and [it specifies that Gurobi is the best solver to use](https://github.com/RobotLocomotion/drake/blob/6609a7e8bcf737f6e0348d4a620e114367511e4d/planning/iris/iris_zo.cc#L362). So I think it is calling Gurobi on this build?

Note that most of the IrisZo tests don't use this -- only tests that set `options.containment_points` require the QP solver. So is there a way to split the tests within IrisZo?

Another thing that could help speed this up is increasing the number of cores that it uses -- the algorithm is designed for heavy CPU parallelism.

rpoyner-tri, 2025-05-07:
I'm not sure parallelism helps the case for memcheck tests at all. Probably better to somehow not run that case under the analyzers.

cohnt, 2025-05-07:
Can we specify no analyzers in some test cases in a program, but not others? Having memory checking for IrisZo has been very helpful in the past, so I'd hate to remove it from all of the tests.

jwnimmer-tri, 2025-05-07:
I think an `args = select(...)` attribute in the BUILD stanza for the test program would be able to pass a `--gtest_filter=...` command line argument that opts-out of specific test names.  (Best would be to rename the tests so that the gurobi-infected ones had a common prefix or something, not listing them one by one.)

cohnt, 2025-05-07:
There are four tests which use it. [One just checks for an appropriate throw, which gets triggered before the solver gets called.](https://github.com/RobotLocomotion/drake/blob/6609a7e8bcf737f6e0348d4a620e114367511e4d/planning/iris/test/iris_zo_test.cc#L299) [The other three are all of the form `TEST_F(FourCornersBoxes, ...)`.](https://github.com/RobotLocomotion/drake/blob/6609a7e8bcf737f6e0348d4a620e114367511e4d/planning/iris/test/iris_zo_test.cc#L380C1-L380C26)

tyler-yankee, 2025-07-01:
7/1 https://drake-jenkins.csail.mit.edu/job/linux-noble-clang-bazel-nightly-everything-valgrind-memcheck/14/

cohnt, 2025-07-14:
Based on @rpoyner-tri's [comments above](https://github.com/RobotLocomotion/drake/issues/22978#issuecomment-2858742191), it sounds like we shouldn't be running any multithreaded code with memcheck. I don't know much about the guts of GTest -- is there a way to run the tests with slightly different parameters depending on the job? If so, we could set it to be single-threaded when running memcheck. (We could also skip the tests that call the commercial solvers, or force them to use an open-source solver.)

cohnt, 2025-07-14:
In one set of test cases, IrisZo solves QPs. If a commercial solver is available, then it will (probably?) be used over the open-source ones. I just bring it up because I know there was discussion if this might be a case, but if Gurobi and Mosek are disabled in the memcheck builds, then it shouldn't be a problem.

I guess I can come back and try to write new test cases at some point for the memcheck builds, but no promises on when that happens.

tyler-yankee, 2025-07-16:
7/16 https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-nightly-everything-release/515/

tyler-yankee, 2025-07-16:
Good catch, thanks! I saw the test name and jumped to conclusions. 🙂