RobotLocomotion/drake #22985: CLP Fails to Solve Simple QP

cohnt, 2025-05-07:
### What happened?

ClpSolver cannot solve the following very simple QP:
```
from pydrake.all import ClpSolver, MathematicalProgram

prog = MathematicalProgram()
x = prog.NewContinuousVariables(2)
prog.AddLinearConstraint(x[0] + x[1] == 1)
prog.AddBoundingBoxConstraint(0, 1, x[1])
prog.AddQuadraticErrorCost(1, [-0.1, 0], x)

solver = ClpSolver()
result = solver.Solve(prog)
print(result.is_success())
```

### Relevant log output

```shell

```

jwnimmer-tri, 2025-05-07:
CLP is excluded from `ChooseBestSolver` for QPs because it tripped a valgrind error (#14933).  The example prints a warning:

```
>>> result = solver.Solve(prog)
WARNING:drake:Currently CLP solver has a memory issue when solving a QP. The user should be aware of this risk.
```

I am not sure if the memory error is still present (we've upgraded CLP since then), but that could be one explanation.

cohnt, 2025-05-07:
Good to know. Should we keep the issue open for now?

jwnimmer-tri, 2025-05-07:
I would say the primary bug is that our documentation says that [CLP cannot solve QPs](https://drake.mit.edu/doxygen_cxx/group__solvers.html) but yet it [claims that it can at runtime](https://github.com/RobotLocomotion/drake/blob/08365f413c4c7e27c79c223a0385206308c8ec25/solvers/clp_solver_common.cc#L36), which seems like it's a consequence of not following up on #14933 and the TODO in #15541.  Until we decide one way or other other whether it's _supposed_ to work, we can't tell whether the expected behavior here should have been "solve it and return true" or "throw an exception that QPs are not supported".

cohnt, 2025-05-07:
I would suggest that this example is strong evidence that CLP cannot solve QPs. Happy to PR a change to make it not claim it can solve QPs at runtime.

(And if so, would it be a breaking change? Bugfix? Would it need to follow the deprecation schedule?)

jwnimmer-tri, 2025-05-07:
On the other hand, CLP has a bunch of QP unit tests in `solvers/test/clp_solvers_test.cc` which _are_ passing.

jwnimmer-tri, 2025-05-07:
I'll use #22988 to see whether the valgrind report is still a problem.

cohnt, 2025-05-07:
Here's a C++ version of the test, that can be added into `solvers/test/clp_solvers_test.cc`. This really is a very simple and obviously feasible QP, so either CLP can't be trusted to solve QPs, or there's a bug somewhere in the parsing.

```
GTEST_TEST(ClpSolverTest, QuadraticProgram22985) {
  MathematicalProgram prog;
  auto x = prog.NewContinuousVariables(2);
  prog.AddLinearConstraint(x[0] + x[1] == 1);
  prog.AddBoundingBoxConstraint(0, 1, x[1]);
  prog.AddQuadraticErrorCost(1, Eigen::Vector2d(-0.1, 0), x);

  ClpSolver solver;
  auto result = solver.Solve(prog, {}, {});
  EXPECT_TRUE(result.is_success());
}
```

sherm1, 2025-05-08:
Assigned to @hongkai-dai for disposition.

cohnt, 2025-07-10:
I've tried to investigate this a little further. Looks like CLP returns the solution status 10, which may be the secondary status described [here](https://github.com/coin-or/Clp/blob/master/src/ClpModel.hpp#L462)?

Regardless, I think we should probably be including a more detailed warning if the user tries to solve a QP with CLP. Right now, we just say
```
WARNING:drake:Currently CLP solver has a memory issue when solving a QP. The user should be aware of this risk.
```
(which seems to no longer be the case per #22988), but says nothing about the fact that it may straight up return the wrong answer. I'm happy to make that PR.

(Really, I think we shouldn't allow solving QPs with CLP at all, but it seems the jury is still out on that question.)

AlexandreAmice, 2025-07-14:
FWIW: [#247](https://github.com/coin-or/Clp/issues/247) claims there is an issue with the presolver when solving QPs in CLP and this program is trivial to pre-solve `x0` away. I haven't actually looked that this is the issue, but I wouldn't be surprised if it is.

cohnt, 2025-07-20:
Thank you for filing the issue upstream!

> BTW, we do have a warning inside ClpSolver when the user attempts to solve a QP with CLP

Yes, I added that warning in #23199.

hongkai-dai, 2025-07-21:
> Yes, I added that warning in https://github.com/RobotLocomotion/drake/pull/23199.

Ah I see, thank you very much for adding the warning, very helpful!

closed 2026-10-03 by sherm1