RobotLocomotion/drake #22996: [ci] Switch primary developer platform from Jammy to Noble

jwnimmer-tri, 2025-05-12:
For developer-centric build flavors (memory checkers, documentation, etc.) we try to run just 1 CI OS flavor, matching the daily driver of the development team's local computers.  Soon, that OS will switch from Jammy to Noble.

To align CI with the developer upgrade, we should change it so that _all_ Jammy builds morph to Noble, and _all_ Noble builds morph to Jammy.  So swap the word `jammy` in the build name out for `noble`, and vice versa.

Not all of the changes need to happen at once.  If it's easier to switch them over in piecemeal, that's fine, but imagine all-at-once will be simplest.

As usual, we want to have `-experimental` flavors available for everything in Production.

Note that the choice of pre-merge required/triggered builds also needs to track this, e.g., they should be a subset of Continuous.  So I suppose all of them should vice-versa as well (not macOS of course).

I will let you know later on when the switch should happen (not yet).  Prior to the switch, we should still test the `-experimental` jobs for any new configurations to make sure they succeed.

tyler-yankee, 2025-05-20:
It might also be worth checking whether we have any Jammy-specific logic in CI, e.g. https://github.com/RobotLocomotion/drake-ci/blob/9fde80c1353132d8e73ade0eef10edcbfc1da785/driver/configurations/cmake.cmake#L248-L250. (I don't _think_ this particular example needs to change? With low confidence ... I just happened to run into this snippet while working on something else.)

jwnimmer-tri, 2025-05-20:
For the purposes of this ticket, I'm okay bending the "swap everything" rule from upthread, and leaving kcov alone if a replacement is not ready by the time of the switch-over.

If we don't already have a separate Drake ticket for "need kcov on noble", please open one now, FYI @rpoyner-tri.

Aiden2244, 2025-05-21:
Couple of updates:
1. I should clarify that the `kcov` issue only affects the coverage jobs. Other issues are causing some other bazel jobs to fail (which I am currently investigating), but this missing dependency is not the root cause there.
2. #23031 is now up!

Aiden2244, 2025-05-22:
### Results of Jammy -> Noble initial conversion
All `linux-jammy` production jobs with either a continuous or nightly flavor were cloned into respective `linux-noble-experimental` jobs. After initial testing, the following jobs were unsuccessful:

#### Sanitizer Jobs
- https://drake-jenkins.csail.mit.edu/view/AddressSanitizer/job/linux-noble-clang-bazel-experimental-address-sanitizer/1/
- https://drake-jenkins.csail.mit.edu/view/AddressSanitizer/job/linux-noble-clang-bazel-experimental-everything-address-sanitizer/1/
- https://drake-jenkins.csail.mit.edu/view/Linux%20Noble/job/linux-noble-clang-bazel-experimental-leak-sanitizer/1/
- https://drake-jenkins.csail.mit.edu/view/Linux%20Noble/job/linux-noble-clang-bazel-experimental-everything-leak-sanitizer/1/
- https://drake-jenkins.csail.mit.edu/view/Linux%20Noble/job/linux-noble-clang-bazel-experimental-thread-sanitizer/1/
- https://drake-jenkins.csail.mit.edu/view/Linux%20Noble/job/linux-noble-clang-bazel-experimental-everything-thread-sanitizer/1/
- https://drake-jenkins.csail.mit.edu/view/Linux%20Noble/job/linux-noble-clang-bazel-experimental-undefined-behavior-sanitizer/1/console
- https://drake-jenkins.csail.mit.edu/view/Linux%20Noble/job/linux-noble-clang-bazel-experimental-everything-undefined-behavior-sanitizer/1/

Sanitizer jobs of all varieties appear to all be failing for a similar reason, with some variation of the following error message:

```bash
/usr/bin/ld.gold: error: cannot open /usr/lib/llvm-15/lib/clang/15.0.7/lib/linux/libclang_rt.lsan-x86_64.a: No such file or directory
```

#### Coverage Jobs
- https://drake-jenkins.csail.mit.edu/view/AddressSanitizer/job/linux-noble-clang-bazel-experimental-address-sanitizer/1
(others untested but presumably fail for the same reason)

Coverage jobs are broken due to #23031

#### Valgrind Memcheck Jobs
- https://drake-jenkins.csail.mit.edu/view/Linux%20Noble/job/linux-noble-clang-bazel-experimental-everything-valgrind-memcheck/1/
- https://drake-jenkins.csail.mit.edu/view/Linux%20Noble/job/linux-noble-clang-bazel-experimental-valgrind-memcheck/1/

Appear to be failing due to a single error: `realloc() with size 0`

#### Documentation Job
- https://drake-jenkins.csail.mit.edu/view/Linux%20Noble%20Unprovisioned/job/linux-noble-unprovisioned-gcc-bazel-experimental-documentation/1/

Appears to be related to Drake's monkeypatch of the `sphinx` plugin at `//docs/pydrake`

#### The good news
All standard jobs, as well as the mirror-to-s3 job, seem to be ok. Furthermore, CI is working as intended for the unstable jobs, even if there are code errors being reported.

Aiden2244, 2025-05-24:
There are only two production Jenkins jobs that are Noble-only at this time:

```
linux-noble-clang-bazel-continuous-release
linux-noble-gcc-bazel-continuous-release
```

Prior to this ticket, the equivalent experimental jobs for Jammy did not exist. [DJJ #181](https://github.com/robotlocomotion/drake-jenkins-jobs/pull/181) created those jobs, both of which passed:

https://drake-jenkins.csail.mit.edu/job/linux-jammy-clang-bazel-experimental-release/1
https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-experimental-release/1

tyler-yankee, 2025-05-26:
Should this ticket also include switching over DEE? (See https://github.com/RobotLocomotion/drake-external-examples/blob/b2a139bc6be587e2104a1183b8411538d70ef478/Jenkinsfile#L19)

jwnimmer-tri, 2025-05-30:
> Sanitizer Jobs

Please re-test all of them after #23054 lands and is baked into new Provisioned images.

> Coverage Jobs

Tracked under #18969.  This need not be a blocker, we can cut over other jobs and let this flavor sync up later.

> Valgrind Memcheck Jobs

TBD

> Documentation Job

Tracked under #23052.  This is probably the highest priority of all of the not-yet-working jobs.

> Should this ticket also include switching over DEE?

Yes, sure.  (Both the GHA jobs and the Jenkins jobs.)

I'm OK for DEE to transition anytime, we don't need to wait for TRI to be ready.

tyler-yankee, 2025-06-03:
Another miscellaneous note on this ticket -- I assume (could be wrong) we'll want to update the documentation here: https://github.com/RobotLocomotion/drake/blob/17383fb009191dd88a6a91e59cbf6e0e3559c869/doc/_pages/from_source.md?plain=1#L51

...and I suppose anywhere else where we reference Jammy as a primary development platform.

jwnimmer-tri, 2025-06-04:
> ... we'll want to update the documentation here: ...

Yup!

tyler-yankee, 2025-06-25:
Closing as we've transitioned DEE over to Noble.