RobotLocomotion/drake #22997: CI: Implement Mac caching

BetsyMcPhail, 2025-05-12:
Follow up to #22858 

Before jumping into the implementation, complete the following:

- [ ] Have a way to measure if this actually speeds things up or slows things down
  - [ ] Needs a baseline measurement ahead of time
- [ ] Wait until Ubuntu caching flakes have stabilized (#21255)
- [ ] Confirm macOS is in the same AZ as the cache server

The goal is to try turning on remote cache similarly to how Ubuntu does it. 

- Same exact server / IP
- Leave the server double size but revisit later
- Certain jobs should populate it (read/write/readwrite). See #19341 for details.

tyler-yankee, 2025-05-15:
Re: baseline (as of writing; this data goes back a couple weeks).

* [mac-arm-sonoma-clang-bazel-continuous-release](https://drake-jenkins.csail.mit.edu/job/mac-arm-sonoma-clang-bazel-continuous-release/) ~37 min
* [mac-arm-sonoma-clang-bazel-nightly-everything-release](https://drake-jenkins.csail.mit.edu/job/mac-arm-sonoma-clang-bazel-nightly-everything-release/) ~47 min.
* [mac-arm-sonoma-clang-cmake-nightly-release](https://drake-jenkins.csail.mit.edu/job/mac-arm-sonoma-clang-cmake-nightly-release/) ~11 min.
* [mac-arm-sonoma-clang-cmake-nightly-everything-release](https://drake-jenkins.csail.mit.edu/job/mac-arm-sonoma-clang-cmake-nightly-everything-release/) ~11min.
* [mac-arm-sonoma-clang-wheel-nightly-release](https://drake-jenkins.csail.mit.edu/job/mac-arm-sonoma-clang-wheel-nightly-release/) ~12min.
* Packaging jobs don't/won't use the remote cache.

Sequoia metrics (considering experimental jobs for the ones that don't have Sequoia production equivalents) are relatively similar.

tyler-yankee, 2025-06-02:
In addition to monitoring if this causes any issues and benchmarking the new build times, we still need to update buildcop and internal documentation. We also may consider cleaning up https://github.com/RobotLocomotion/drake-ci/blob/main/cache_server/health_check.bash as its first (and only) argument is now meaningless.

tyler-yankee, 2025-06-10:
AFAIK, all scripts, jobs, and documentation (in `drake/doc` and internal) have been updated to only refer to a single cache server, and we haven't had any issues with macOS jobs using the cache. Any future issues can go to their own tickets.