RobotLocomotion/drake #23016: CI: Review and improve S3 Glacier rules

BetsyMcPhail, 2025-05-15:
Early this week, doxygen was unintentionally moved to glacier when it reached the 5 year timeout. 

The purpose of this issue is to study current glacier rules and offer suggestions on improvements.

Related to past work in https://github.com/RobotLocomotion/drake/issues/17435

jwnimmer-tri, 2025-05-15:
Scribbing down the one thought I had, not sure if it's right -- possibly the main problem here is hosting doxygen on drake-packges instead of drake-mirrors like all of our other externals.

tyler-yankee, 2025-06-23:
@jwnimmer-tri A few questions for you as we've started looking into this one.

1. drake-apt: We still have releases in here from Ubuntu Xenial, Bionic, and Focal. Should we keeping these around even though they're no longer officially supported?
2. VTK: We have _a lot_ of miscellaneous copies of VTK in various places, e.g. drake-homebrew, drake-provisioning, and multiple in drake-packages. I'd like to have a better sense of what actually gets used so that we can document it better and clean up anything that's stale.

As far as the Doxygen move to drake-mirror, I tend to agree with your assertion above, but if provided a compelling reason I could be convinced otherwise. It turns out we actually _do_ already have a copy of Doxygen in `drake-packages/other/doxygen`. Regardless, I'd still need to look into what needs to change if we get rid of the one in packages. 

There's some other miscellaneous cleanup/organization we can do while we're in here, but I think those are the major outstanding items.

tyler-yankee, 2025-06-24:
Well, it seems like Drake is already using Doxygen from multiple places.

From drake-mirror:
https://github.com/RobotLocomotion/drake/blob/8c3a249184ed09f0faab3c678536d66d732809ce/tools/workspace/doxygen/Dockerfile#L27

From drake-packages:
https://github.com/RobotLocomotion/drake/blob/8c3a249184ed09f0faab3c678536d66d732809ce/tools/workspace/mirrors.bzl#L26-L29

It's not clear to me what the difference between these two / which are used / what the purpose of each is?

jwnimmer-tri, 2025-06-30:
> drake-apt: We still have releases in here from Ubuntu Xenial, Bionic, and Focal. Should we keeping these around even though they're no longer officially supported?

As with our other packages (in `drake-packages`), strictly speaking we only need to keep them alive for 5 years after their release date, and after that move them into glacier.  However, I imagine it might be difficult to automate given how the apt site works.  If it's difficult to make it happen, it's probably fine to just leave it alone.

> VTK: We have a lot of miscellaneous copies of VTK in various places, e.g. drake-homebrew, drake-provisioning, and multiple in drake-packages. I'd like to have a better sense of what actually gets used so that we can document it better and clean up anything that's stale.

See https://github.com/svenevs/drake/blob/6076b315827f035b56c064e8416bcd324fe5e8a4/tools/workspace/vtk/README.md for a (now deleted) file about VTK retention (search "5 years").  That pre-dates our current design of building VTK from source.

The general idea is we need to keep any of our "rebuild" artifacts for 5 years after they were last used.  So to answer this precisely, you need to go back and figure out when they were last used.  It's probably worth a some effort on that front, but the bail-out answer is to find when we started using github VTK source builds.  It would have been after #20068 (which added some rules, but didn't switch to using them), perhaps #19945.  You could add 5 years after that date for expiry (so around 2028 or somthing).

> It's not clear to me what the difference between these two / which are used / what the purpose of each is?

The "source" one is the source code.  CI maintainers run that Dockerfile to compile new binaries.

The second one (drake-packages) is the Doxygen binaries, which is what our build actually downloads and runs.

tyler-yankee, 2025-07-09:
> difficult to automate given how the apt site works

What do you mean by this? Could we not have the same glacier rules "as with our other packages (in `drake-packages`)"?

I'll have to do some digging on the VTK piece, but thanks for the head start.

BetsyMcPhail, 2025-07-11:
Would using tags be possible? https://docs.aws.amazon.com/AmazonS3/latest/userguide/troubleshoot-lifecycle.html#troubleshoot-lifecycle-8

tyler-yankee, 2025-07-15:
> Would using tags be possible?

After looking into it, tags still don't get around the fundamental limitation that you can't apply 'exclusion' criteria, only 'inclusion'.

If we were to just have multiple lifecycle rules, with one saying "keep the htmls" and the other saying "delete all old files, regardless of size", I tend to think that the [S3 conflict policies](https://docs.aws.amazon.com/AmazonS3/latest/userguide/lifecycle-conflicts.html) state it would delete the files we don't want it to:

> Generally, Amazon S3 Lifecycle optimizes for cost. ... Permanent deletion takes precedence over transition.

However, [object locks](https://docs.aws.amazon.com/AmazonS3/latest/userguide/object-lock.html) could work. From [here](https://docs.aws.amazon.com/AmazonS3/latest/userguide/object-lock-managing.html#object-lock-managing-lifecycle):

> Object lifecycle management configurations continue to function normally on protected objects, including placing delete markers. However, a locked version of an object cannot be deleted by a S3 Lifecycle expiration policy. Object Lock is maintained regardless of which storage class the object resides in and throughout S3 Lifecycle transitions between storage classes.

We would just 'lock' the html files we want to keep indefinitely. ([Here's](https://docs.aws.amazon.com/AmazonS3/latest/userguide/object-lock-configure.html) a guide on setting them up.)

BetsyMcPhail, 2025-07-16:
The object locks look promising! Nice find

tyler-yankee, 2025-07-17:
I've added the locks to the html files and modified the lifecycle rules in drake-packages to have a minimum size of 0. There was a warning message about it possibly incurring greater cost to transition these files to Glacier / delete them than to keep them around, but I don't see this being anything substantial. The [pricing page](https://aws.amazon.com/s3/pricing/) indicates the lifecycle transition request to Glacier Flexible is $.03/1000 requests.

The remaining tasks here are to (a) check back in ~24 hours and make sure that it's working as intended, i.e. the old sha's are gone but the ones within the last five years are still there, and (b) to add this to our internal documentation.

tyler-yankee, 2025-07-18:
> The general idea is we need to keep any of our "rebuild" artifacts for 5 years after they were last used. ... You could add 5 years after that date for expiry (so around 2028 or something.

Okay. The lifecycle rules are already set up this way in drake-packages; there are old very old copies (2018,2019) which are currently in Glacier and the newest version here (9.2.0) would be set to expire July 2028, 5 years from when it was created. Should we be Glacier-ing those old versions on expiry or fully deleting them?

When we make that decision, I also need to update the lifecyle rules for drake-homebrew's VTK, which has no expiry applied (i.e., we still have old files available in standard storage), to match drake-packages.

jwnimmer-tri, 2025-07-18:
For example, above I mentioned #19945 which was the date at which `https://drake-packages.csail.mit.edu/vtk/vtk-9.1.0-3-focal-x86_64.tar.gz` stopped being used.  It's fine to use the date of that commit (2023-09-07) instead of the next release (2023-09-14) as the baseline.  They are within rounding error anyway when we're talking about keeping intact for 5 years (alive in S3 until >= 2028-09-07).

tyler-yankee, 2025-07-21:
I checked this morning re: the object locks .... the old sha's are gone, but so is the index.html file in drake/nightly despite putting a lock on it. To me this directly contradicts the statement from their documentation that I quoted above (https://github.com/RobotLocomotion/drake/issues/23016#issuecomment-3074954580), but it doesn't look we have much choice.

I will reupload the file for now. The only other solution I can come up with is that all the files we actually care about in drake/nightly are prefixed with `drake`, so we could apply the lifecycle rule to `drake/nightly/drake` instead of just `drake/nightly/` and still keep the non-`drake`-prefixed html files around.

tyler-yankee, 2025-07-21:
Re: VTK in drake-packages and drake-homebrew

> Note that "used" != "created". ...

Yes. However that's unfortunately not really how S3 lifecycle rules work, they are limited based on the day of creation. The best I can come up with here is to get rid of our current Glacier rule for VTK (since we won't be uploading any more files there, unless something changes on that front) and by-hand add "object metadata" which sets the expiry to a custom date per-object based on time of last use plus five years that I trace back in drake's history.

Also, @jwnimmer-tri would you know we we have VTK 9.2 release candidate binaries in drake-packages? #19945 only references 9.1, and if it was built from source afterwards I don't see how newer versions would come into the picture.

tyler-yankee, 2025-08-13:
With the drake-packages lifecycle rules updated to

1. be more consistent and apply to time of deprecation rather than time of creation for VTK
2. clean up the old SHA files that weren't being deleted, without removing the necessary HTML files

and the investigation into the remaining buckets complete, the remaining work here is to set a reminder somewhere to Glacier or remove drake-homebrew after 5 years of it no longer being in use by drake. That was when #21454 landed on 2024-05-24, so 2029-05-24. Then, #23280 just has to be completed within that timeframe.