RobotLocomotion/drake #23025: symbolic mass matrix (and friends) have broken

RussTedrake, 2025-05-18:
### What happened?

The symbolic manipulator equations in my underactuated intro chapter notes broke with Drake v1.41.  

I've produced a minimal c++ repro, and run `git bisect` and localized the change to #22977. 

For the reproduction, I added the following to `multibody_plant_test.cc`:
```
TEST_F(SplitPendulum, Symbolic) {
    auto sym_plant = systems::System<double>::ToSymbolic(plant_);
    auto sym_context = sym_plant->CreateDefaultContext();

    // State variables
    Eigen::VectorX<symbolic::Expression> q =
        symbolic::MakeVectorVariable(1, "q");
    Eigen::VectorX<symbolic::Expression> v =
        symbolic::MakeVectorVariable(1, "v");
    Eigen::VectorX<symbolic::Expression> vd =
        symbolic::MakeVectorVariable(1, "vd");

    // Parameters
    Eigen::VectorX<symbolic::Expression> m =
        symbolic::MakeVectorVariable(2, "m");
    Eigen::VectorX<symbolic::Expression> l =
        symbolic::MakeVectorVariable(2, "l");

    const auto& upper_arm = sym_plant->GetBodyByName("upper_section");
    SpatialInertia<symbolic::Expression> inertia0(
        m[0],
        Vector3<symbolic::Expression>(0, 0, -l[0]),
        UnitInertia<symbolic::Expression>(l[0] * l[0], l[0] * l[0], 0)
    );
    upper_arm.SetSpatialInertiaInBodyFrame(sym_context.get(), inertia0);
    const auto& lower_arm = sym_plant->GetBodyByName("lower_section");
    SpatialInertia<symbolic::Expression> inertia1(
        m[1],
        Vector3<symbolic::Expression>(0, 0, -l[1]),
        UnitInertia<symbolic::Expression>(l[1] * l[1], l[1] * l[1], 0)
    );
    lower_arm.SetSpatialInertiaInBodyFrame(sym_context.get(), inertia1);
    // const auto& elbow_frame = sym_plant->GetJointByName("elbow").frame_on_parent();
    // elbow_frame.SetPoseInParentFrame(
    //     sym_context.get(), RigidTransform<symbolic::Expression>([0, 0, -l[0]])
    // );

    sym_plant->SetPositions(sym_context.get(), q);
    sym_plant->SetVelocities(sym_context.get(), v);
    Eigen::MatrixX<symbolic::Expression> M(1,1);
    sym_plant->CalcMassMatrixViaInverseDynamics(*sym_context, &M);
    Eigen::VectorX<symbolic::Expression> Cv(1), tauExt(1);
    sym_plant->CalcBiasTerm(*sym_context, &Cv);
    Eigen::VectorX<symbolic::Expression> tauG =
        sym_plant->CalcGravityGeneralizedForces(*sym_context);
    Eigen::MatrixX<symbolic::Expression> B = sym_plant->MakeActuationMatrix();
    EXPECT_EQ(B.rows(), 1);
    EXPECT_EQ(B.cols(), 1);
    MultibodyForces<symbolic::Expression> forces(*sym_plant);
    sym_plant->CalcForceElementsContribution(*sym_context, &forces);
    sym_plant->CalcGeneralizedForces(*sym_context, forces, &tauExt);
    tauExt -= tauG;
}
```
This passes before the change, but now fails with 
```
unknown file: Failure
C++ exception with description "The following environment does not have an entry for the variable m(0)

" thrown in the test body.
```

@sherm1 -- as the author of #22977, I'll assign this to you.

### Version

source

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell

```

sherm1, 2025-05-19:
Yikes -- I guess we don't have a unit test for this. Will investigate today.

sherm1, 2025-05-19:
@RussTedrake please review PR #23027 asap

PR #23027 (2025-05-20): Fix symbolic mass matrix computation
PR #22977 added a precalculation for body spatial inertias that broke some symbolic calculations (it did some numerical checks for NaNs and zeroes). This PR limits those checks to numerical scalars, and adds a missing unit test (thanks, Russ!) that will prevent recurrence.

Resolves #23025

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/23027)