RobotLocomotion/drake #23031: [build] Remedy missing kcov package for Noble

Aiden2244, 2025-05-21:
See #22996 for context.

As it would turn out, the `kcov` package is entirely missing from Noble's `apt` universe (despite being included in later releases, see [this discussion](https://answers.launchpad.net/ubuntu/+source/kcov/+question/818841)). This is causing the newly minted [noble coverage Jenkins jobs](https://drake-jenkins.csail.mit.edu/view/Coverage/job/linux-noble-clang-bazel-experimental-coverage/) to fail, as that workflow depends entirely on `kcov`.

In order to address this, we should:
1. Build `kcov` from source as part of the source distribution `install_prereqs.sh` workflow on noble.
2. Change the coverage workflow so that it no longer depends on `kcov`.
3. Come up with some clever solution I have not yet thought of.

jwnimmer-tri, 2025-05-21:
This might be a duplicate of #18969.  WDYT @rpoyner-tri?

rpoyner-tri, 2025-05-22:
It's a near-dup of the other issue. In a sense, noble is doing us a favor by forcing the issue.

Late-model kcov sources have updated support for debian packaging (that should have made it through the normal maintainer pipeline? not sure what held things up), so if need be it shouldn't be too hard to host a docker-build-to-deb somewhere, and have drake consume the deb.