RobotLocomotion/drake #23044: [ci] Streamline XCode base image setup

tyler-yankee, 2025-05-27:
Currently, our process for creating a new XCode base image is very cumbersome. In particular, we'd like to:

1. Download an Xcode installer, which we host from Kitware, rather than having to download manually from Apple's website and transfer the file over to the image.
2. Eliminate the need to screen share to change some settings, replacing with command-line solutions.

We have a bunch of ideas for how to do some of the steps involved here sitting in internal docs, but creating a ticket to track progress. This can be tested the next time we have to make a new XCode base image, and the code for (2) should likely live in a script on drake-ci similar to [Ubuntu's `ami_init_script`](https://github.com/RobotLocomotion/drake-ci/blob/main/setup/ubuntu/ami_init_script).

mathstuf, 2025-05-27:
Note that our Xcode installers are not publicly hosted (I really don't know what the license terms are on that, but I can't imagine it is "trivial"). We have Ansible rules that we use to deploy them to machines (with support of sending them from the host directly to the target so you're not slinging the huge installer out and back over the VPN). I imagine some way to slurp from S3 instead can be done (I'll defer any using-AWS-smartly decisions as I'm not really familiar).

I can extract the commands done in Ansible for use here.

tyler-yankee, 2025-08-19:
Once https://github.com/RobotLocomotion/drake-ci/pull/340 lands, the remaining work is to:

* update internal documentation to remove any code that was previously pasted there and is now in drake-ci
* decide on / write issues for any follow-ups (e.g., downloading the Xcode and CLT files automatically if we can get the bot account auth figured out)

closed 2025-09-02 by tyler-yankee