RobotLocomotion/drake #23053: [ci] June 2025 provisioned image refresh

tyler-yankee, 2025-05-30:
Last month: #22930 

* [ ] Linux Jammy
* [ ] Linux Noble
* [ ] macOS Sonoma
* [ ] macOS Sequoia

macOS Sequoia also needs to update to XCode 16.4.

tyler-yankee, 2025-05-30:
@jwnimmer-tri Do we expect any other traffic to come in related to #22996? We could possibly wait another week or two before doing the Noble image?

jwnimmer-tri, 2025-05-30:
I'm not sure.  However, we will need to re-bake for #23054 right away anyway (so we can unveil any further errors), so I would say just do the monthly refresh steps as part of that one.  If we discover additional Noble changes beyond that, we can bake more images later on.

tyler-yankee, 2025-06-04:
All images have been updated, and documentation was updated above.

closed 2025-06-04 by tyler-yankee