RobotLocomotion/drake #23067: Switch release engineering to Noble

tyler-yankee, 2025-06-04:
Spawned from #22996. When Kitware changes over development platforms to Noble to match TRI, we'll be running the release scripts from that OS version. We currently document in several places across `tools/release_engineering` that only Jammy is supported. We need to investigate whether the current tooling works on Noble, and update the documentation accordingly.

tyler-yankee, 2025-06-10:
Looks like we won't be able to make this upgrade internally for another couple months, so this ticket might sit for a while. (We could still do the investigation in some container, but it's probably not worth the time investment.)

jwnimmer-tri, 2025-09-22:
Per f2f -- all seems to work correctly, just need some docs updates.