RobotLocomotion/drake #23107: `//geometry/render_gl:internal_opengl_context_test causes` ASAN failure

Aiden2244, 2025-06-17:
Brief background: since #23066 and #23078 landed, the production jobs that were previously run exclusively on Jammy now run on Noble (see [DJJ #189](https://github.com/robotlocomotion/drake-jenkins-jobs/pull/189)).

One new job, `linux-noble-clang-bazel-continuous-everything-address-sanitizer`, is failing due to a memory leak in this issue's titular test, the output from which is summarized below:

```
[11:26:20 AM]  FAIL: //geometry/render_gl:internal_opengl_context_test (Exit 1) (see /media/ephemeral0/ubuntu/workspace/linux-noble-clang-bazel-continuous-everything-address-sanitizer/_bazel_ubuntu/f2af90fde150cbd7df9098204f6f7e08/execroot/_main/bazel-out/k8-dbg/testlogs/geometry/render_gl/internal_opengl_context_test/test.log)
[11:26:20 AM]  [15,876 / 17,679] 241 / 995 tests, 1 failed; Testing //systems/analysis:implicit_euler_integrator_test (shard 1 of 2); 26s remote-cache, processwrapper-sandbox ... (7 actions running)

...

[11:26:20 AM]  [==========] 5 tests from 2 test suites ran. (4332 ms total)
[11:26:20 AM]  [  PASSED  ] 5 tests.
[11:26:20 AM]  
[11:26:20 AM]  =================================================================
[11:26:20 AM]  ==4032085==ERROR: LeakSanitizer: detected memory leaks
[11:26:20 AM]  
[11:26:20 AM]  Direct leak of 2352 byte(s) in 49 object(s) allocated from:
[11:26:20 AM]      #0 0x5555556b8072 in __interceptor_malloc (/media/ephemeral0/ubuntu/workspace/linux-noble-clang-bazel-continuous-everything-address-sanitizer/_bazel_ubuntu/f2af90fde150cbd7df9098204f6f7e08/execroot/_main/bazel-out/k8-dbg/bin/geometry/render_gl/internal_opengl_context_test+0x164072) (BuildId: 1bdac84d60cb9309b0386910cea622996d4a14d0)

...

[11:26:20 AM]      #29 0x55555581899d in testing::Test::Run() external/+drake_dep_repositories+gtest/googletest/src/gtest.cc:2739:5
[11:26:20 AM]  
[11:26:20 AM]  SUMMARY: AddressSanitizer: 2352 byte(s) leaked in 49 allocation(s).
```

Please see the following job for more info:
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-noble-clang-bazel-continuous-everything-address-sanitizer/1/

Aiden2244, 2025-06-17:
Closing for now in the event this was a fluke, will reopen if https://drake-jenkins.csail.mit.edu/job/linux-noble-clang-bazel-continuous-everything-address-sanitizer/2/ has the same issue.

Aiden2244, 2025-06-17:
> will reopen if https://drake-jenkins.csail.mit.edu/job/linux-noble-clang-bazel-continuous-everything-address-sanitizer/2/ has the same issue.

Turns out this one passed, so this is perhaps a fluke.

jwnimmer-tri, 2025-06-18:
For now let's mark this 1 test as "no_asan" with a TODO citing this issue number, and keep this issue open to track resolving it more seriously later on.

jwnimmer-tri, 2025-09-04:
This is tricky, because it doesn't (usually) reproduce locally, because it depends on using the same MESA drivers as CI, but locally I have (for example) the NVidia drivers, and other will have whatever drivers their desktop needs.  To repro locally, someone will probably need to figure out how to switch GL to MESA in a local test/build.  Or else debug on an EC2 machine provisioned the same as CI.

tyler-yankee, 2025-09-30:
Thanks, but after running it twice with that flag (so 2000 total) still nothing.

I'm not inclined to just re-enable the tests though, because it failed multiple times above and nothing else has changed in `geometry/render_gl` since.

tyler-yankee, 2025-11-05:
Can't reproduce with a for-loop with 100 iterations wrapping the entire test body.

tyler-yankee, 2025-11-05:
I guess resetting revealed the underlying pattern, which is that the leak only ever occurs the very first time the test runs on an image? (Same behavior on a new instance, minus the `mesa-utils` install altogether.)

jwnimmer-tri, 2025-11-05:
Ah, okay.  If it happens just the first time, then it won't accumulate memory, so we can suppress it without worry.