RobotLocomotion/drake #23123: Updated Ubuntu Noble AMIs (mold linker)

rpoyner-tri, 2025-06-23:
In #22951 , the `install_prereqs` changed, for Ubuntu Noble *only*. Please re-bake new Ubuntu Noble Provisioned AMIs for CI and deploy them.

tyler-yankee, 2025-06-26:
@jwnimmer-tri @rpoyner-tri The new Noble provisioned image has the prereqs from master, 22951 (mold), and 23124 (clang-19), so nothing should be breaking to transition to the new images.

tyler-yankee, 2025-07-30:
@rpoyner-tri Since #22951 finally landed and then got reverted, I'd like to clarify for August's new images. Could you confirm that doing:

```
apt-get install mold
```

on the Noble provisioned image is all that is needed on top of the installation prereqs currently on master?

jwnimmer-tri, 2025-07-30:
Yes, that's correct.